LibreCAD/LibreCAD #1089: Hang in RS_Line::draw when opening dxf file

mycae, 2019-04-25:
Librecad hangs when opening a DXF file I have, particularly when running the RS_Line::draw() routine. I'm running 2.1.2 from Debian stretch ( 2.1.2-1+b1). 

Specifically, in my file (which I can share privately),  when opening the dxf the program hangs in the below loop (rs_line.cpp), as length is very large and total never reaches it:
```

84         RS_Vector curP{pStart+direction*total};
685         for (int j=0; total<length; j=(j+1)%pat->num) {
686 
687         // line segment (otherwise space segment)
688                 double const t2=total+fabs(ds[j]);
689                 RS_Vector const& p3=curP+dp[j];
690         if (ds[j]>0.0 && t2 > 0.0) {
691             // drop the whole pattern segment line, for ds[i]<0:
692             // trim end points of pattern segment line to line
693                         RS_Vector const& p1 =(total > -0.5)?curP:pStart;
694                         RS_Vector const& p2 =(t2 < length+0.5)?p3:pEnd;
695             painter->drawLine(p1,p2);
696         }
```
```
(gdb) info locals
t2 = 1310637897.8169217
p3 = @0x7fffffffa4f0: {x = 1969.3615593626603, y = 1310638740.816921, z = 0, valid = true}
j = 0
curP = {x = 1969.3615593626603, y = 1310638733.2629642, z = 0, valid = true}
patternSegmentLength = 3
total = 1310637897.8169217
viewportRect = {_minP = {x = -1494.8223371730753, y = -807.2950248931669, z = 0, valid = true}, _maxP = {x = -300.93532852825098, y = -168.2812164566038, z = 0, valid = true}}
endPoints = {vector = std::vector of length 2, capacity 2 = {{x = -2.0000000000000031, y = -88, z = 0, valid = true}, {x = -2.000000000000004, y = -90, z = 0, valid = true}}, tangent = false}
ec = {<RS_Entity> = {<RS_Undoable> = {<RS_Flags> = {_vptr.RS_Flags = 0x555555e85648 <vtable for RS_EntityContainer+16>, flags = 2}, cycle = 0x0}, parent = 0x0, minV = {x = 10000000000, y = 10000000000, z = 0, 
      valid = true}, maxV = {x = -10000000000, y = -10000000000, z = 0, valid = true}, layer = 0x0, id = 44205, pen = {<RS_Flags> = {_vptr.RS_Flags = 0x555555e82888 <vtable for RS_Pen+16>, flags = 0}, 
      lineType = RS2::SolidLine, width = RS2::Width00, screenWidth = 0, color = {<QColor> = {cspec = QColor::Rgb, ct = {argb = {alpha = 65535, red = 0, green = 0, blue = 0, pad = 0}, ahsv = {alpha = 65535, 
              hue = 0, saturation = 0, value = 0, pad = 0}, acmyk = {alpha = 65535, cyan = 0, magenta = 0, yellow = 0, black = 0}, ahsl = {alpha = 65535, hue = 0, saturation = 0, lightness = 0, pad = 0}, 
            array = {65535, 0, 0, 0, 0}}}, <RS_Flags> = {_vptr.RS_Flags = 0x555555e82868 <vtable for RS_Color+16>, flags = 0}, <No data fields>}}, updateEnabled = true, varList = std::map with 0 elements}, 
  entities = {<QListSpecialMethods<RS_Entity*>> = {<No data fields>}, {p = {static shared_null = {ref = {atomic = {_q_value = {<std::__atomic_base<int>> = {static _S_alignment = 4, 
                  _M_i = -1}, <No data fields>}}}, alloc = 0, begin = 0, end = 0, array = {0x0}}, d = 0x555556b52300}, d = 0x555556b52300}}, subContainer = 0x0, static autoUpdateBorders = false, entIdx = -1, 
  autoDelete = true}
pStart = {x = 1969.3615593626603, y = 843, z = 0, valid = true}
pEnd = {x = 1575, y = 8.880232858974633e+17, z = 0, valid = true}
direction = {x = -4.4408920985006257e-16, y = 1, z = 0, valid = true}
length = 8.880232858974624e+17
pat = 0x555555eafd00 <RS_LineTypePattern::patternDashLineTiny>
pen = {<RS_Flags> = {_vptr.RS_Flags = 0x555555e82888 <vtable for RS_Pen+16>, flags = 0}, lineType = RS2::SolidLine, width = RS2::Width00, screenWidth = 0, color = {<QColor> = {cspec = QColor::Rgb, ct = {
        argb = {alpha = 65535, red = 65535, green = 65535, blue = 65535, pad = 0}, ahsv = {alpha = 65535, hue = 65535, saturation = 65535, value = 65535, pad = 0}, acmyk = {alpha = 65535, cyan = 65535, 
          magenta = 65535, yellow = 65535, black = 0}, ahsl = {alpha = 65535, hue = 65535, saturation = 65535, lightness = 65535, pad = 0}, array = {65535, 65535, 65535, 65535, 0}}}, <RS_Flags> = {
      _vptr.RS_Flags = 0x555555e82868 <vtable for RS_Color+16>, flags = 0}, <No data fields>}}

```
Note that the length variable is >1e17, however the end and start points when entering the function in the debugger seemed sane.

As a hack. I've attached a suggested patch to circumvent this by checking the size of length and bailing out 

- as its still compiling, I've not tested this. Nor have I worked out why length is very large.

Interestingly, when I disable the construction setting on the layer, the problem goes away and I can view my file normally.  I can also manually set in the debugger length to, say, 100, and then program continues (only one line in the drawing)

mycae, 2019-04-25:
OK, github wont let me upload patches (unsupported filetype). here it is:
```
diff -r c813af237309 rs_line.cpp
--- a/rs_line.cpp	Thu Apr 25 21:28:47 2019 +0100
+++ b/rs_line.cpp	Thu Apr 25 21:40:23 2019 +0100
@@ -645,6 +645,10 @@
         return;
     }
 //    patternOffset = remainder(patternOffset - length-0.5*pat->totalLength,pat->totalLength)+0.5*pat->totalLength;
+    const float MAX_LENGTH=1e9;
+    if(length > MAX_LENGTH)
+	    return;
+
     if(length<=RS_TOLERANCE){
         painter->drawLine(pStart,pEnd);
         return; //avoid division by zero
```

lordofbikes, 2019-04-25:
There are a couple of similar cases collected in #915.
Most, if not all, are solved in latest code.
So you can try to build latest code and see if it works or send your file to me, then I can check if it works with latest version.

mycae, 2019-04-25:
Ive finished compiling the latest git head, and yes the problem does appear to be solved - sorry for the noise!

closed 2019-04-25 by mycae

lordofbikes, 2019-04-26:
No reason to feel sorry, you're welcome!
And many thanks for testing this.