LibreCAD/LibreCAD #1090: Feature Request: Filter selection by properties

nickajeglin, 2019-04-25:
Hello, 

I'd like to request a command to filter selected entities by property. Basically a filter command that gives a dialog box that allows boolean logic to select or exclude items from the selection set. For example, select a bunch of lines, circles, arcs, etc., use the command to exclude all lines that are the color green, on layer x, and are horizontal. 

This is something that I find really useful when working in very complicated drawings with many overlapping features. 

See also draftsight's "qselect" feature.

shawncurry, 2019-05-24:
What other sorts of filter criteria do you have in mind?  What types of operations would you normally perform on these selections?

nickajeglin, 2019-05-24:
Here's two example use cases:
1) I have a DXF for a part that contains several different hole sizes, but many of them are 10.5mm to mate with a weld nut. A laser operator notices that the cutting bur is reducing the effective size of the hole, so I need to change any hole that is 10.5mm to 10.75mm without changing the other, unrelated, holes. I'd like to be able to select everything, pick circle from a dropdown, choose diameter from another dropdown, and then type 10.5 into a textbox to select only the relevant holes.

2) I have a DXF of a flattened sheet metal part exported from creo or some other parametric 3D drafting software. The laser machine needs a single closed contour, but creo often includes doubled lines that are related to how the material stretches at a bend relief. These doubled lines (or polylines or splines) are a different color than the main contour. I'd like to be able to select all, then include or exclude based on color. 

After selection, ideally I'd be able to change the properties of the entity as a group. For the first selection set I'd change all diameters to 10.75, and for the second, I'd just delete them or move to a different layer. 

So for filter criteria, I'm thinking of a dropdown that lists all types of entities currently in the selection set, and a second dropdown that normally lists any properties that are common to all entities (layer, color, line type, etc), and dynamically includes properties that are specific to the type of entity entered in the first box (diameter, length, horizontal, vertical, angle, etc.)

shawncurry, 2019-05-30:
Thanks @nickajeglin, it's much clearer now.  This does sound like a great feature.