LibreCAD/LibreCAD #1098: Add a preference to allow whitespace in command interface

shawncurry, 2019-05-12:
As an EU of the LibreCAD command interface, I prefer the old 2.1.3 method which allowed me to use white space in my commands.

### Expected behavior

I can use white space.

### Observed behavior

Typing a space causes immediate evaluation before I was ready.

### Steps to reproduce or sample file

Use LibreCAD 2.2 rc1

### Operating System and LibreCAD version info

Win10, built from master, using Qt5.12.3/MinGW7.3, also Visual Studio 2017

This behavior was added for Issue #764.  It appears to have been designed to mimic AutoCAD functionality.  This should probably have a preference; I think it will be confusing for new users who are inexperienced with AutoCAD,

shawncurry, 2019-05-12:
This could be a check item in the quick options list:

[image]

The caption could probably use some work - suggestions welcome.

vwren, 2019-05-13:
Sorry I'm late to the party.  Once I thought about it, I realized the message box could be improved.  I left a comment in the commit, if that's kosher.

lordofbikes, 2019-05-14:
Discussion moved to #1099

shawncurry, 2019-05-19:
This preference should be moved into the Application Preferences dialog.

commit 0a35309e (2019-06-10): - fix #1098
- add new checkable button and preference to allow whitespace in the command interface (the 2.1.3 behavior)
closed 2019-06-10 by lordofbikes