LibreCAD/LibreCAD #1100: Cannot open files which contain Unicode characters in the file name

shawncurry, 2019-05-13:
As an international LibreCAD EU, I want to open my files which contain unicode characters in the file name.

### Expected behavior

The file opens.

### Observed behavior

[image]

### Steps to reproduce or sample file

Ａマンション25d.dxf

### Operating System and LibreCAD version info

Win10
LibreCAD 2.1.3
LibreCAD current master; Qt5.12.3/MinGW7.3 / Visual Studio 2017

I hate QString may need a mention in my bio.  Thanks Qt for including both a toStdString and toStdWString; now I have to make sure every usage which deals with files will handle UTF-16.

shawncurry, 2019-05-13:
Suggestion: "lc_char.h"

`pseudo-code:
namespace lc {
#ifdef _UNICODE
typedef wchar_t lc_char;
typedef std::wstring lc_string;
#define toLCString() toStdWString()
#else
typedef char_t lc_char;
typedef std::string lc_string;
#define toLCString() toStdString()
#endif
}
`

etc.  Then use it of course.

shawncurry, 2019-05-19:
This is probably os specific, but I've only tested on windows.  Armin @lordofbikes, can you confirm if this is working on other os?

xanderdin, 2019-05-19:
On my Debian Linux system I have no problems saving or opening with specified filename.

fa201, 2019-05-19:
I tested on Xubuntu (debian-based) and I am able to save and open a file with the name you supplied.
Version: 2.2.0-rc1-142-ge38b7088
Compiler: GNU GCC 7.4.0
Compiled on: May 18 2019
Qt Version: 5.9.5
Boost Version: 1.65.1
System: Ubuntu 18.04.2 LTS

I can also import on 2.1.3.