LibreCAD/LibreCAD #1105: Support AutoCAD SHX fonts

shawncurry, 2019-05-17:
As an EU, I want my drawings containing SHX fonts to be displayed correctly.

### Expected behavior

I can see text that uses SHX fonts.

### Observed behavior

The text does not appear.

### Steps to reproduce or sample file

Open a DXF which uses an SHX font:
[Edited.zip](https://github.com/LibreCAD/LibreCAD/files/3193051/Edited.zip)

### Operating System and LibreCAD version info

Win10
LibreCAD 2.1.3

I know, I know, SHX is a proprietary format.  SHP, however, which is the uncompiled version, are plain text and the standard is well documented by AutoDesk.  SHP versions of the most widely used SHX fonts are widely available.  The information contained in them amounts to the geometry needed to render the font, for each character.

One could therefore simulate SHX support, by implementing a text renderer which uses a corresponding SHP to determine how to stroke them out.

shawncurry, 2019-05-18:
This is on the must-have list.  I looked through github and a couple other places and didn't really find anything specific to fonts.  The reader part is simple enough, but fonts should also have a renderer that transforms the geometry according to the STYLE:

> AutoCAD is packaged with numerous text fonts. You can use the STYLE command to apply expansion, compression, or obliquing to any of these fonts, thereby tailoring the characters to your needs. You can draw text of any height, at any baseline angle, and with either horizontal or vertical orientation using these fonts.

Link to the standard: [http://docs.autodesk.com/ACD/2011/ENU/filesACG/WS73099cc142f4875513fb5cd10c4aa30d6b-7f3c.htm](url)

shawncurry, 2019-05-18:
Of course the easy way out of having to support them natively would be to attempt a lookup of an lff font to use as an alternative.

shawncurry, 2019-05-18:
It looks like there could be a few more user stories to shake out of this one.  

One approach could be to provide a corresponding shptolff conversion utility, and a couple of preferences:

1. Whether to attempt a lookup of the LFF equivalent if the requested font is not an LFF.
2. Whether to preserve or strip out the text if the LFF cannot be found.

Another approach would be to actually provide a renderer for SHP fonts as described above.  This shouldn't look too much different from the LFF renderer, as the formats appear to be very similar.

shawncurry, 2019-05-22:
As an EU, I would like to manage all of my Font preferences in one place.

**Acceptance Criteria**

- A new tab has been added to Application Preferences for Font preferences
- The font folder paths are managed on this page
- A function is available on this page to select and convert TTF and SHX fonts to the LFF format
- Converted fonts are placed into the appropriate folder path
- The user may select whether to perform a font alias lookup on this page
- A list of font alias names can be managed on this page
- The user may choose whether to strip all text referencing unknown fonts, or preserve it, undisplayed

lordofbikes, 2019-05-22:
I agree that the support of various font formats will need some preferences the user must be able to adjust.
But I can imagine, that a font converter GUI may overload a preferences tab.
Having a separate widget for the GUI allows to call it from various places, e.g. from menu, preferences or text tools.
Also consider an Unix like solution. We can keep and improve standalone _ttf2lff_ and _whatever2lff_ converter and make a nice GUI inside LibreCAD for them.
This will work for all OS of course and also allow batch conversions, e.g. for automated deployment.

shawncurry, 2019-05-22:
Hi Armin, 

I like the idea of having a separate UI widget to automate the command line utilities.  You are right, this would be useful in other areas as well, such as the Text input dialogs.  Here comes another user story:

As an End User, I would like a graphical utility so I can more easily convert my fonts to the LFF format.

**Acceptance Criteria**
The utility can launch a "File Open" dialog in order to select multiple font files for conversion.
The default file filter includes file types supported for each available command line conversion utility
The utility uses the command line tools to perform the conversions.
The produced LFF fonts are placed into the user's selected font folder in Application Preferences

shawncurry, 2019-05-24:
Lack of support for TTF may also be the cause of a crash in dwg import.  Reference Issue #988

scls19fr, 2019-11-09:
I don't know if that's related to this issue but noticed a huge difference in font rendering between LibreCAD and QCad... 

I think this screenshot can help to understand what I mean
[image]

app4soft, 2024-08-22:
SHX2LFF converter listed here:

- https://github.com/LibreCAD/LibreCAD/issues/1244#issuecomment-2295087090