LibreCAD/LibreCAD #1108: Improve User Experience while working with multiple drawings

shawncurry, 2019-05-23:
As an End User who often works with multiple drawings open, I want to have my drawing window settings remembered, and to be able to save all my changes with one click.

### Expected behavior

The program remembers my preferences and provides a way to "Save All" changes.

### Observed behavior

It does not.

### Steps to reproduce or sample file

- Open multiple drawings
- Choose Drawings -> Tab mode
- Click Zoom Extents
- Close LibreCAD  <-- prompts to save each file
- Reopen LibreCAD  <-- returns to Sub-window mode, with windows maximized

### Operating System and LibreCAD version info

Win 10
Version: 2.2.0-rc1
Compiler: GNU GCC 4.9.1
Compiled on: Feb 16 2018
Qt Version: 5.4.1
Boost Version: 1.53.0

shawncurry, 2019-05-23:
Let's face it - the Drawings menu is clunky.  When you switch between tab and sub-window, the menu choices change.  The "feel" of this menu could benefit from some checkable menu items, and possibly sub-menus.

The Save dialog that appears when you close the program, could possibly use a "Do this for the next <n> items" checkbox.  It also has a dated look.

A "Save All" button in the toolbar and File menu would also be handy.

shawncurry, 2019-05-23:
The auto-save behavior should also be reviewed for this case. 

#1086 improve autosave for unnamed files

shawncurry, 2019-05-25:
Actions

File -->
Close
Close All
Save unnamed1
Save unnamed1 As...
Save All

pseudo-code:
```
save() {
  if (dirty && showSaveAsIfNeeded(currentFileName; default="") == canceled)
    return false;
  return true;
}

saveAll() {
  foreach (document) {
    if (save() == false)
      return false;
  }
  return true;
}
```

etc

shawncurry, 2019-05-25:
- Tab names should not include the path; only the file name
- Full path can be displayed in the hover hint

shawncurry, 2019-06-10:
[image]
[image]
[image]
[image]

closed 2019-06-10 by lordofbikes

lordofbikes, 2019-06-10:
What I wonder is, how **rounded** looks like in Windows.
On Linux **rounded** tabs are rather rectangular
[image]

shawncurry, 2019-06-11:
Yes, the "Rounded" description seems misleading at best: (lol)
[image]

I prefer the "Triangular" version, which looks similar to the south-position tabs in several windows applications, like Excel.  The rectangular ones look "old" and ugly to me.

PR #1122 (2019-06-10): Feature/issue1108
- **Fix #1108 Improve User Experience while working with multiple drawings**
- Fix #515 DWG zoom position on opening
- Fix #1085 Zoom panning can only be used once and must then be react
- Add File > Save All and File > Close All actions + icons
- This commit also includes my un-merged changes from #1099 (preferences stuff)
commit 52652be8 (2019-06-10): - Fix #1108 Improve User Experience while working with multiple drawings
- Fix #515 DWG zoom position on opening