LibreCAD/LibreCAD #1109: Improve Undo capabilities

shawncurry, 2019-05-26:
As an End User, I want to be able to undo and redo more types of actions.

### Expected behavior

I can undo the renaming of a block.

### Observed behavior

When I try to undo this change, nothing happens.  The only actions I can undo are simple additions and removal of entities.  No "state change" actions can be undone.

### Steps to reproduce or sample file

Draw a circle.
Select the circle and create a new block.
Rename the block.
Click undo/redo.

### Operating System and LibreCAD version info

Version: "2.2.0-alpha"
Compiler: Microsoft Visual C++
Compiled on: May 26 2019
Qt Version: 5.12.3
Boost Version: 1.70.0
System: Windows 10 (10.0)

shawncurry, 2019-05-26:
Suggested solution: RS_UndoableAction

pseudo-code:

```
class RS_UndoableAction : public RS_Undoable {
  virtual void doUndoableAction() = 0;  // do,,, the un-doable
  virtual void undoUndoableAction() = 0;  // wait, nevermind
  virtual void undoStateChanged(bool undone) override {
    if (undone)
      undoUndoableAction();
    else
      doUndoableAction();
  }
}

class RS_RenameBlockAction : public RS_UndoableAction {
  // . . . 
  void doUndoableAction() {
    block.setName(newName);
  }
  void undoUndoableAction() {
    block.setName(oldName);
  }
}
```

shawncurry, 2019-05-26:
I have successfully completed a significant refactoring and redesign of the RS_Undoable classes.  I actually started this work while investigating Issue #716 "Removed blocks are persistent somewhere".

I introduced the classes discussed above, and a couple more.  I also successfully used the new classes to implement undo/redo the naming of a block.  Here is a brief overview my my commit notes.

- Extracted RS_Undo implementation from RS_Document; now it has an RS_Undo, instead of being one
- Introduce RS_UndoManager as an implementation of RS_Undo, also as notification center for undo events
- Introduce RS_UndoListener, RS_Document now uses this interface to be notified of changes (instead of performing them)
- Introduce RS_UndoableAction as a solution for undoing actions other than add/remove entity
- Introduce RS_BlockRenameAction as an example of an undoable action

The problem discussed by @Equoda still exists at the moment, but I'm close to having that fixed as well.  This is an unintended behavior owing to a bit of a design shortcut; likely taken in part because of the previous lack of a listener/notification center system for undo events.  Undone blocks will now actually be removed from the block list, and tranferred to a separate "undone" list.

There will be a lot more work to be done with the new undo capabilities.  In the previous design, the only type of action that could be undone was adding or removing an Entity, which does cover a lot of ground.  

Using the new classes, any type of state changes can now be easily undone as well, like renaming, freezing, visibility changes, etc.  One needs only to implement the RS_UndoableAction interface, and then they will be able to do was previously thought impossible.  💯

Symbian9, 2023-02-27:
> I have successfully completed a significant refactoring

Was there a pull request somewhere?

Latest LibreCAD 2.2.0 release still has issue with persisting "removed blocks":

- https://github.com/LibreCAD/LibreCAD/issues/716