LibreCAD/LibreCAD #1110: Grip points, a small improvement

susnicek, 2019-05-27:
Dear Librecad developers,
first of all I would like to thank you for the effort you are investing in Librecad development. It is a great piece of software, which can easily substitute the Drafsight 2D.
Let me suggest a small improvement:

If I select an object (e.g a line) I can move or modify it using dragging grip points (those little blue squares that appear when you select something). It would be nice if the new position of a such dragged point can be specified in the command window (e.g. by a command @100<0).

Best regards, Susnicek

### Operating System and LibreCAD version info
Version: 2.2.0-alpha (compiled on 24th May)

shawncurry, 2019-05-30:
Hi @susnicek ,

I think what you're asking for might already be available with the Modify > Move/Copy command.  If you had something a little different in mind, please reply and let us know.

- Select an entity
- Type "move"
- Specify a reference point (can be selected with the mouse or typed into the command window)
- Specify target point (mouse or command window again. Incremental "@" input is accepted)

susnicek, 2019-05-30:
Dear shawncurry,

thank you very much for your reply. I am aware that moving an entity can be done by the "move" command. But what I mean is a little bit different thing. Consider this example:
Let's say that I have just plotted a simple line. Now, when I select the line, the line style changes to a dashed style and two small blue squares appear at the both ends of the line. Let's call the small blue squares the "grip points" - I think it is the AutoCad therminology.
I found out that it is possible to drag each grip point of the line (using mouse move with pressed left button) and move it to a different place, while the second line grip point remains at its spot.
However, the new position of the grip point can be selected only by the mouse and it is not possible to prescribe the grip point new position using a command. My feature request was to supply the posiblity of placement of the grip point also using the standard librecad commands.

Best regards, Susnicek

shawncurry, 2019-06-05:
I see what you are saying.  Selecting one of the grip points, allows you to move it with the mouse, however command input is not available in this mode.  This sounds like a practical request to me.  

I'd like to learn a little more about the situation.  When you are drawing the line in the first place, the command input is available.  Are these existing lines, that need to be adjusted then?  Or is it a case of mistaken input?  I ask because, I have been experimenting with a redesign of the undo, so that more things could be undone easily, perhaps even mistaken commands.  But that probably won't help much if you are fixing an existing drawing.

susnicek, 2019-06-05:
Dear shawncurry,
it not a case of mistaken input command. It is going on the case, when I have an existing item (e.g. a line) and want to modify it (e.g. move one of its point somewhere else). If I am moving the grip point, the QtLineEdit for command input is enabled and it is possible to write a command in it. If I do so, the command has no effect on position of my grip point. The only way for placement of the grip point is  a mouse click.

I tried to debug the code and found out that my command during positioning of a grip point is recognized as command, however is, from some reason, not performed. Probably such action is not registered in this mode. The code, which I see first time, is for me too complex and cannot add this feature.

Best regards, Susnicek