LibreCAD/LibreCAD #1115: Enter to Command interface with special character (@)

tochimoreno, 2019-06-05:
### Expected behavior
press @ char to enter to the command interface to input relative coordinates to move/copy/create shapes

### Observed behavior
It doesn't catch the @ key in my keyboard layout so it doesn't enter to the command interface.

### Steps to reproduce or sample file
With a LATAM or ES keyboard layout.
press l, "SAPCEBAR" to create a line
click for the first point
press "Alt Gr" + Q to enter @ nothing happend
press Q, to enter to enter to the command interface, delete that Q and now press "Alt Gr" + Q to enter the @ charater

### Operating System and LibreCAD version info
Ubuntu 18.04.2 LTS
LibreCad 2.2.0-alpha
Keyoard layout: LATAM or ES

lordofbikes, 2019-06-05:
This seems to be a duplicate of #1016, which was solved in Oct 2018.
LibreCAD 2.2.0-alpha is not latest code. If you build from source, you should checkout a newer version.

tochimoreno, 2019-06-06:
Sorry for the duplicate, I've read a lot of issues looking for this and I haven't found it.
2.2.0-alpha is the last in the dev repo for ubuntu. If there is another way to install a newer version (without have to compile it) please let mi know.

Thank you for the response.

fa201, 2019-06-06:
You may check the build instruction for Ubuntu, which are quite simple, at : https://librecad.readthedocs.io/en/latest/appx/build.html#building-on-linux

fa201, 2020-04-13:
@lordofbikes 
Should we close this issue ?

hfuerst, 2023-11-02:
This problem still exists !!!
@ is not recognized

LibreCad installed Today:
Version: 2.2.0.2
Compiler: GNU GCC 13.1.0
Erstellt am: Jul 29 2023
Qt Version: 5.12.11
Boost Version: 1.75.0
System: Windows 10 (10.0)

dongxuli-concord-design, 2023-11-02:
Do you also have problem with entering a space? I thinking about to support 1 1/2"

lordofbikes, 2023-11-02:
See actual _Application Preferences > Defaults > Keyboard Settings_, there are already alternative functions for `SPACE BAR`. Allowing `SPACE` for imperial values, will collide and can cause issues if we don't implement a kind of XOR logic for _Keyboard Settings_.

Anyhow, this issue is about getting focus to the command line. Usually, the command line is activated by any key press. My fix from #1016 seems to work no longer for `@` or there is a different cause.

To reproduce:
1. start 2 Point line tool
2. click for the starting point (command line has no focus then)
3. a) press a number key to enter absolute coordinate -> command line gets focus
   b) press `@` -> command line does not get focus to enter relative coordinates

Workarounds are:
- use any letter or number key to focus command line and delete it with `BACKSPACE`
- use shortcut `CTRL`+`m` to focus command line

tochimoreno, 2023-11-02:
> This problem still exists !!! @ is not recognized
> 
> LibreCad installed Today: Version: 2.2.0.2 Compiler: GNU GCC 13.1.0 Erstellt am: Jul 29 2023 Qt Version: 5.12.11 Boost Version: 1.75.0 System: Windows 10 (10.0)

I use the double point method for relative coordinates.
Enter X offset, enter .., enter Y offset.
100..50

It is much simpler than the @ method and works always.