LibreCAD/LibreCAD #1116: Command-line from file does not show

yiyuezhuo, 2019-06-05:
I try to find a method to read command from file instead of typing them line by line and find this post

http://forum.librecad.org/Command-line-from-file-td5716351.html

It claim nighly version(https://sourceforge.net/projects/librecad/files/Windows/NightlyBuilds/master/
) provide a button right beside the command line. But I can't find it even in this version. Though I can find it in other screenshot from other post such as this one:

### Expected behavior
http://forum.librecad.org/quot-Key-Mode-quot-quot-Load-Command-File-quot-quot-Paste-Multiple-Commands-quot-td5716611.html

[image]

### Observed behavior

But I can't find it here:

[image]

It seems that all of those textboxes exceed the expected right boundry, is it just a UI bug? Can I fix it by some dragging?

### Steps to reproduce or sample file

Open and find it.

### Operating System and LibreCAD version info

Version 2.1.3

lordofbikes, 2019-06-05:
For me, on Linux, and latest code the button is always there, even when I resize the window to an unusable size.
So it would be good to know which OS you use and which exact LibreCAD version (see Help -> About).

As a work around, you can simply move the command widget and place it below the drawing area, where it will have much more space.
You can drag the widget when you click and hold the left mouse button on the *Command line' header and move it around.

yiyuezhuo, 2019-06-06:
Oops, It seems that I download 2.1.3 instead of 2.2.0 which is required to show this windows...

closed 2019-06-06 by yiyuezhuo