LibreCAD/LibreCAD #1118: LibreCAD 2.2 rc1 crashes when opening a file on the command line

shawncurry, 2019-06-06:
### Expected behavior

From the command line, I can type:  LibreCAD.exe "C:\some\path\to\drawing.dxf", and it will open the file.

### Observed behavior

It crashes.

### Steps to reproduce or sample file

See above

### Operating System and LibreCAD version info

Version: "2.2.0-alpha"
Compiler: Microsoft Visual C++
Compiled on: Jun  6 2019
Qt Version: 5.12.3
Boost Version: 1.70.0
System: Windows 10 (10.0)

shawncurry, 2019-06-06:
I logged qc_applicationwindow.cpp, and I suspect #877 was the cause of this regression.

closed 2019-06-07 by shawncurry

lordofbikes, 2019-06-07:
Because #877 was a Mac OS specific issue, I'm still concerned about this.
Also in #877 LibreCAD opened an empty drawing but didn't crash.
Maybe it's related with the DXF file itself?
I can't remember any issues with that on Windows. Basically it's the same like opening a DXF file in *Windows Explorer*, then the DXF path is also applied by command line argument.
Perhaps it was an issue with the current working folder or environment? In a development environment the LibreCAD.exe may have pulled wrong Qt libs or something like this.

shawncurry, 2019-06-07:
Hi Armin @lordofbikes,

I'm sorry, I jumped too quickly in reporting this as a bug.  It was a setup issue on my end.  I'll try to be more careful in the future.