LibreCAD/LibreCAD #1126: translation contribure

saro1992, 2019-06-18:
im new learner of librecad,i want contribute to chinese translation,i found some translation error on my debian sid(librecad ver 2.1.3),some word are still english(translation not complete).i register account to translate.librecad.org and found the ts file version arent the latest(some new word are unsearchable),and in github i21n branch have some New translations but  the some  translation  almost are new . what shoud i do.

saro1992, 2019-06-18:
i compiled the lastest version (master branch)of the librecad .
~~i21n~~>l10n

DuncanLithgow, 2021-04-16:
Did you find out how to contribute to translations? We'd love to have your contribution.

The proper place for this discussion is on the forum here: https://forum.librecad.org/LibreCAD-translate-f5710645.html

DuncanLithgow, 2021-04-19:
@lordofbikes what is the issue here which is now pending? Is there a plan to get translate.librecad.org working again?

fa201, 2023-08-09:
@lordofbikes updated the wiki to share the new translating plaftorm: https://dokuwiki.librecad.org/doku.php/community:contribute_translate

DuncanLithgow, 2023-10-04:
@saro1992 I'm going to assume that the information on the wiki is an answer to your question. I'll close this issue for now.

closed 2023-10-04 by DuncanLithgow