LibreCAD/LibreCAD #1130: errors during build related to lifetime of arrays

xzcvczx, 2019-07-01:
seems to be related/due to #1111 (or maybe #1125) using clang on macOS

```
actions/rs_actiondrawcircletan2.cpp:49:26: error: backing array for
      'std::initializer_list' member 'enTypeList' is a temporary object whose
      lifetime would be shorter than the lifetime of the constructed object
RS_ActionDrawCircleTan2::RS_ActionDrawCircleTan2(
                         ^~~~~~~~~~~~~~~~~~~~~~~
actions/rs_actiondrawcircletan2.h:82:50: note: initializing field 'enTypeList'
      with default member initializer
  ...enTypeList {RS2::EntityLine, RS2::EntityArc, RS2::EntityCircle};
     ^          ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from actions/rs_actiondrawcircletan3.cpp:27:
In file included from lib/gui/rs_dialogfactory.h:31:
In file included from lib/gui/rs_dialogfactoryadapter.h:32:
In file included from lib/engine/rs_block.h:31:

actions/rs_actiondrawcircletan1_2p.cpp:53:29: error: backing array for
      'std::initializer_list' member 'enTypeList' is a temporary object whose
      lifetime would be shorter than the lifetime of the constructed object
RS_ActionDrawCircleTan1_2P::RS_ActionDrawCircleTan1_2P(
                            ^~~~~~~~~~~~~~~~~~~~~~~~~~
actions/rs_actiondrawcircletan1_2p.h:84:50: note: initializing field
      'enTypeList' with default member initializer
  ...enTypeList {RS2::EntityLine, RS2::EntityArc, RS2::EntityCircle};
     ^          ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
actions/rs_actiondrawcircletan2_1p.cpp:53:29: error: backing array for
      'std::initializer_list' member 'enTypeList' is a temporary object whose
      lifetime would be shorter than the lifetime of the constructed object
RS_ActionDrawCircleTan2_1P::RS_ActionDrawCircleTan2_1P(
                            ^~~~~~~~~~~~~~~~~~~~~~~~~~
actions/rs_actiondrawcircletan2_1p.h:78:50: note: initializing field
      'enTypeList' with default member initializer
  ...enTypeList {RS2::EntityLine, RS2::EntityArc, RS2::EntityCircle};
     ^          ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

actions/rs_actiondrawcircletan3.cpp:55:26: error: backing array for
      'std::initializer_list' member 'enTypeList' is a temporary object whose
      lifetime would be shorter than the lifetime of the constructed object
RS_ActionDrawCircleTan3::RS_ActionDrawCircleTan3(
                         ^~~~~~~~~~~~~~~~~~~~~~~
actions/rs_actiondrawcircletan3.h:82:50: note: initializing field 'enTypeList'
      with default member initializer
  ...enTypeList {RS2::EntityArc, RS2::EntityCircle, RS2::EntityLine, RS2::EntityPoint};
     ^          ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```

### Expected behavior

### Observed behavior

### Steps to reproduce or sample file
scripts/build-osx.sh

### Operating System and LibreCAD version info
macOS 10.14.4; libreCAD master

feinerer, 2019-08-03:
This also happens on OpenBSD 6.5-current (amd64) using clang 8.0.0 (the default compiler).

sand1024, 2026-09-26:
could be closed I suppose

closed 2026-09-26 by sand1024