LibreCAD/LibreCAD #1132: PPA support for Ubuntu 18.x and 19.x

pgporada, 2019-07-04:
### Expected behavior
Installation on an Ubuntu 18.x and 19.x can't be done securely according to the example message below. Can someone poke upstream for PPA support on these distros?

### Observed behavior
```
$ sudo add-apt-repository ppa:librecad-dev/librecad-daily
 
 More info: https://launchpad.net/~librecad-dev/+archive/ubuntu/librecad-daily
Press [ENTER] to continue or Ctrl-c to cancel adding it.

Hit:1 http://ppa.launchpad.net/certbot/certbot/ubuntu disco InRelease
Hit:2 http://archive.canonical.com/ubuntu disco InRelease                                                                                               
Hit:3 http://debian.sur5r.net/i3 disco InRelease                                                                                                                     
Hit:4 http://us.archive.ubuntu.com/ubuntu disco InRelease                                                                                                            
Hit:5 http://us.archive.ubuntu.com/ubuntu disco-updates InRelease                                                                                             
Get:6 http://us.archive.ubuntu.com/ubuntu disco-backports InRelease [88.8 kB]                                                                    
Hit:7 http://security.ubuntu.com/ubuntu disco-security InRelease                                                               
Ign:8 http://ppa.launchpad.net/librecad-dev/librecad-daily/ubuntu disco InRelease                                              
Hit:10 http://ppa.launchpad.net/system76-dev/stable/ubuntu disco InRelease                                                           
Err:11 http://ppa.launchpad.net/librecad-dev/librecad-daily/ubuntu disco Release                 
  404  Not Found [IP: 91.189.95.83 80]
Hit:9 https://packagecloud.io/slacktechnologies/slack/debian jessie InRelease
Reading package lists... Done                                  
E: The repository 'http://ppa.launchpad.net/librecad-dev/librecad-daily/ubuntu disco Release' does not have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
```

### Operating System and LibreCAD version info
```
$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 19.04
Release:	19.04
Codename:	disco
```

fa201, 2019-07-04:
The issue appeared a few months ago on 18.04 LTS. An alternative is to build from source. It is quite easy once the proper packages have been installed.

pgporada, 2019-07-04:
Hi,
I ended up doing just that and was able to eventually make some basic wood
joins!

jakimfett, 2020-01-09:
Observing this behaviour on Elementary OS v5.1 (Ubuntu 18.04.3 LTS) as well.

I want to use this software, not have to compile it though, @fa201.

Any chance of getting a fix?
It sounds like...it literally just needs a specific file to exist on your PPA?

fa201, 2020-01-09:
I don't know if this can be easily fixed... Building from source is the only work around I have to get up-to-date version.

jakimfett, 2020-01-20:
> I don't know if this can be easily fixed... 

...the easy fix is to set up the PPA with a release file, as I stated in my previous comment.  
Maybe do a quick re-read of what has been said so far, eh?

The transition to secure apt clients means that options to access insecure PPAs are going away.  
Thus, it is the responsibility of the package maintainer to fix this.

Any idea who that package maintainer would be, @fa201 ?

---
> Building from source is the only work around I have to get up-to-date version.

The point that I think you're missing is,  
(and I'm going to emphasize this since I already said it once)  
### **Users of this software want to _USE_ the software**
_(rather than spend time trying to get it to compile.)_

Building from source is *not* a workaround to get the PPA working securely.  
As such, comments about "you can just compile" are off topic, and you might consider refraining from mentioning that again in this thread.

lordofbikes, 2021-01-24:
To put it straight, this is about missing PPA maintainer, which is tracked in #894

closed 2021-01-24 by lordofbikes