LibreCAD/LibreCAD #1133: an Advice is not correct

cwvb, 2019-07-12:
### Expected behavior

### Observed behavior

### Steps to reproduce or sample file

### Operating System and LibreCAD version info
on 
https://github.com/LibreCAD/LibreCAD/wiki/Build-from-source#macos

you find this:

MacPorts

install MacPorts from http://www.macports.org/

You can install LibreCad using MacPorts by: $ sudo port install librecad

You can build LibreCAD manually by following steps:

Install QT and a new gcc, which should have a version 4.4 or later.

Install a version of Qt and boost, for example $ sudo port install gcc46 **qt5-creator** qt5 boost

> correct port name is  **qt5-qtcreator** 

> $ sudo port install gcc46 **qt5-qtcreator** qt5 

> 

> 

Select the right compiler, as LibreCAD doesn't build with the default llvm-gcc42, $ sudo port select --set gcc mp-gcc46

Unzip or checkout a version of LibreCAD into a directory.

cd LibreCAD
./scripts/build-osx.sh

This creates an executable "LibreCAD.app/Contents/MacOS/LibreCAD" and package "LibreCAD.dmg".
Pages 13

    Home
    Becoming a developer
    Build from source
    Custom Widgets
    Download
    Features
    Git and GitHub
    GitHub Labels
    How to translate LibreCAD
    LFF Font Format
    QC_ApplicationWindow
    Style Sheets
    Widgets

Clone this wiki locally

fa201, 2019-07-13:
Thanks for the feed-back. 
If my understanding is correct, on the WIP [documentation](https://librecad.readthedocs.io/en/latest/appx/build.html#building-on-macos), we should make the following changes on packages:
**qt5-creator-mac** -> **qt5-qtcreator**
Should we also change : **qt5-mac** -> **qt5**
Then we should add the following to complete the build: `./scripts/build-osx.sh` ?

GSStnb, 2019-07-13:
The build instructions on RtD doesn't use MacPorts.