LibreCAD/LibreCAD #1144: Poor visibility of UI when using dark UI theme

Zmuogs, 2019-09-22:
I using KDE Plasma 5.16.5 in Linux. When UI theme is dark (f.e. "Breeze Dark" in this case) then  most of icons, some labels in LibreCAD is poorly visible (black on dark background). See attached pictures.
 UI should adapt to dark UI theme.

[image]
[image]
[image]
[image]

LibreCAD:
Version: 2.1.3
Compiler: GNU GCC 8.2.0
Compiled on: Sep 17 2018
Qt Version: 5.12.2
Boost Version: 1.67.0
System: Ubuntu 19.04

Zmuogs, 2019-09-22:
Note on black label "Command:" in attached screenshot: when LibreCAD is started fresh then that label is white, but mouseclick in edit field of that "Command:" ("Commant:" turns blus) and then hit ESC -> "Command:" turns black.

natrius, 2020-01-24:
Jumping in, its hard to see anything. I'm using "Starcraft" and, well...

[image]

I tried to search for a setting to change the theme just for this program back to breeze default so i can work with it, but i could not find anything. I think another program had it, but unfortunately i cannot remember :(

MikePryadko, 2020-02-04:
I have the same situation:

[image]
```
Version: 2.1.2
Compiler: GNU GCC 6.2.0
Compiled on: Sep 17 2016
Qt Version: 5.9.5
Boost Version: 1.62.0
System: Ubuntu 18.04.4 LTS
```
My DE is Xfce (theme `adwaita-dark`), so I use `qt5-style-plugins` for applying GTK theme to Qt

kevinsmia1939, 2020-03-06:
[image]

Same issue here on KDE Plasma 5
LibreCAD 2.1.3
Operating System: openSUSE Tumbleweed 20200227
KDE Plasma Version: 5.18.1
KDE Frameworks Version: 5.67.0
Qt Version: 5.14.1
Kernel Version: 5.5.6-1-default
OS Type: 64-bit
Processors: 8 × Intel® Core™ i7-3770 CPU @ 3.40GHz
Memory: 7.5 GiB of RAM

deric, 2020-10-26:
A workaround:

1. Install `qt5ct`
Debian/Ubuntu: `sudo apt install qt5ct`

2. Export ENV variable and run `librecad`
```
$ QT_QPA_PLATFORMTHEME=qt5ct librecad
```
3. Go to `Options > Widget Options` and set Style to `qt5-ct-style`.

4. Run `qt5ct` in order to adjust Qt5 theme.

[image]

To persist the settings, add this line to `~/.profile` and logout (and after starting new sessions qt5ct theme should work withou exporting ENV variable):
```
export QT_QPA_PLATFORMTHEME=qt5ct
```

Proper solution would require adding new set of [icons](https://github.com/LibreCAD/LibreCAD/tree/master/librecad/res/icons). And supporting e.g. Breeze Dark

deric, 2020-10-28:
When you go to `Options > Widget Options > Toolbar` and check "Use Themed Icons" you should see that some icons are adjusted to your theme:

[image]

Some icons can be loaded from desktop theme, the set of icons is limited and given by the [freedesktop icons specification](https://specifications.freedesktop.org/icon-naming-spec/icon-naming-spec-latest.html):
```cpp
    if (using_theme)
        action->setIcon(QIcon::fromTheme("document-new", QIcon(":/icons/new.svg")));
    else
        action->setIcon(QIcon(":/icons/new.svg"));
```

LibreCAD icons could be easily converted to dark theme icons using simple bash script. Just go to icons directory:
```
cd librecad/librecad/res/icons
```
and run this script to replace black color in SVG files:
```sh
#!/bin/sh
# convert light colors to dark-theme colors

find -name "*.svg" | while read i;
do
        fname=$( basename "$i")
        fdir=$( dirname "$i")
        sed -i -e 's/#000000/#f2f2f2/g' "$i"
done
```
And [recompile LibreCAD](https://github.com/LibreCAD/LibreCAD/wiki/Build-from-source).

 This way you'll get an UI that looks more usable, although not ideal:
[image]

In order to make this work, the code needs to be refactored. Replace:

```cpp
action->setIcon(QIcon(":/icons/select_intersected_entities.svg"));
```
with something like:
```cpp
    if (using_theme)
        action->setIcon(QIcon::fromTheme("select-intersected-entities", QIcon(":/icons/select_intersected_entities.svg")));
    else
        action->setIcon(QIcon(":/icons/select_intersected_entities.svg"));
```
- create default icons theme
- add `index.theme` file
- reorganize icons directory
- create dark theme icons

Anyone willing to help with this?

Nikita-Presnov, 2022-12-06:
Is it life?

Symbian9, 2023-02-27:
> LibreCAD icons could be easily converted to dark theme icons using simple bash script.  Just go to icons directory and run this script to replace black color in SVG file

As icons are SVG, and parsed by Qt during each launch, could it be solved internally by dynamically replacing colors during parsing icons SVG files? Then there would be no needs to create separate "dark icons" set.

- https://stackoverflow.com/questions/15123544/change-the-color-of-an-svg-in-qt
  - https://forum.qt.io/topic/112405/qt-svg-dynamic-modification
  - https://doc.qt.io/qt-5/qtxml-index.html

It should be implemented as an option "Icons: `Light` / `Dark`" (drop-down toggle) into "Widget Options" window, and take effects on-the-fly (or after restart app).

flywire, 2023-10-14:
https://wiki.inkscape.org/wiki/Creating_a_multicolor_icon

fa201, 2023-10-14:
For record purpose, I add also the [Forum issue](https://forum.librecad.org/Backround-color-td5721547.html)

mangelba, 2024-01-07:
I modified the icons so that they look good in the dark theme. Maybe it will be useful to someone.
https://github.com/mangelba/LibreCAD-with-dark-icons/tree/dark-icons/librecad/res/icons

flywire, 2024-01-07:
What did you change? How about a screen shot of the two sets a comparison?

The editor changed every icon even where there is no change to the appearance of the icon.

mangelba, 2024-01-07:
> What did you change? How about a screen shot of the two sets a comparison?
> 
> The editor changed every icon even where there is no change to the appearance of the icon.

With dark icons:
[image]

Original icons:
[image]

MikePryadko, 2024-01-14:
@mangelba 
> I modified the icons so that they look good in the dark theme. Maybe it will be useful to someone. https://github.com/mangelba/LibreCAD-with-dark-icons/tree/dark-icons/librecad/res/icons

Looks good!
Sorry for dummy question, but how regular users should use your icons?) (how to install it?)

mangelba, 2024-01-23:
> Looks good! Sorry for dummy question, but how regular users should use your icons?) (how to install it?)

It is not easy.  It is necessary to compile the source code.

First download de source code cloning  from github:
$  git clone  https://github.com/mangelba/LibreCAD-with-dark-icons.git
and  go to the dark-icons brank:
$ git checkout dark-icons

Or directly  downlad in dark-icons brank:
$ git clone  --branch dark-icons https://github.com/mangelba/LibreCAD-with-dark-icons.git

and then compile Librecad: https://github.com/LibreCAD/LibreCAD/wiki/Build-from-source

for me in linux works:
$ qmake -r
$ make

I have seen that I will also have to edit the icons that are in png format.

Sorry if I can't explain it well but my knowledge of git is very little (and of the English language too).

kevinsmia1939, 2024-07-05:
@mangelba 
Hi, could you add this to feature into LibreCAD directly? Does the other developers receptive of the change?

lordofbikes, 2024-07-05:
@kevinsmia1939 the above discussion is about a quick'n'dirty solution. The source's icons are modified to compile a application binary for dark theme. But this binary will have dark theme icons on light theme too, thus no practical solution to integrate into the main repo.
It needs a solution, where the end user can switch the whole icon theme at run time. This needs much more efforts than just modifying the icons.

MikePryadko, 2024-07-05:
@kevinsmia1939  does anybody still using light theme in 2024???) I think that that persons simple doesn't know about existence of dark themes)))

flywire, 2024-07-05:
> does anybody still using light theme in 2024

Yes. Depends on conditions, indoor/outdoor, personal preference, OS etc. Anyway, it's irrelevant since ultimately it needs to be selectable.

> With dark icons

@mangelba I think you misunderstood my comment _What did you change?_ Clearly it is different but many of the icons appear the same. Unfortunately iirc your commit obfuscates the icon changes.

zentus, 2024-07-18:
I feel bad for architects having to learn to build a C program from source before even being able to see the icons properly.

flywire, 2024-07-18:
> I feel bad for architects having to learn to build a C program from source before even being able to see the icons properly.

Very misinformed comment given the [workflow](https://github.com/LibreCAD/LibreCAD/blob/master/.github/workflows/build-all.yml) is provided to build executables, eg https://github.com/LibreCAD/LibreCAD/issues/1606#issuecomment-1760541369.

Are you offering to code your suggested solution?

flywire, 2024-07-18:
Your thoughts are aligned with the dev position:

> It needs a solution, where the end user can switch the whole icon theme at run time. This needs much more efforts than just modifying the icons.

Unfortunately, developer resources are limited but they have been able to help others with their coding effort. I'm not up to coding this app but I could provide other support in a Windows environment if it is useful.

dongxuli-concord-design, 2024-07-18:
@zentus 

If would like to try coding, please feel free to get help from us. 
The early learning curve is a little steep. I suggest to have a look at our commit history at GitHub, and you can see most of the coding is extremely simple. The changes also give some ideas on where and what to look.

zentus, 2024-07-20:
> @zentus
> 
> If would like to try coding, please feel free to get help from us. The early learning curve is a little steep. I suggest to have a look at our commit history at GitHub, and you can see most of the coding is extremely simple. The changes also give some ideas on where and what to look.

Great, I'll create a fork.

What would be good entry points in the code base to start looking at:
- Icons import and use in GUI
- Stylesheets set/get and defaults
- Application Preferences GUI
- Application Preferences storage (persisted values)

The PNG icons seem to be generated from SVG files. I'm thinking we could create a script that generates packs (Light and Dark) of SVGs based on SVG templates and theme JSON files. Then use those packs to generate the PNGs, using `imagemagick` or similar. This script could run at build time. We could also include the pregenerated PNGs in the repo, as they currently are.

OS dark mode needs to be detected at runtime. I'm on macOS. The OS dark mode setting is stored as `Dark` in `AppleInterfaceStyle` in `~/Library/Preferences/.GlobalPreferences.plist`. I can get it via terminal command `defaults read -g AppleInterfaceStyle`. We could either read the `.plist` file directly or create a subprocess for the terminal command. 

I'm not sure what is the case on other OS:s. I have a Windows and Ubuntu machine that I could test on. 

For Ubuntu we could create a list of common OS themes considered to be "dark", and check if any of those are currently active, or try to do some magic with looking at background colors via GTK bindings. For Windows 10> there seems to be a registry value `HKEY_CURRENT_USER\Software\Microsoft\Windows\CurrentVersion\Themes\Personalize\AppsUseLightTheme`.

flywire, 2024-07-20:
SVG Icons should cover it. I standardised the pallete in beadb6e and d3985b6 and don't overlook comments in (mostly closed) duplicate issues.

- Firstly, icon set need to be loaded at runtime
- This immediately allows users to change the icon set
- Dark theme is a misnomer because there are all sorts of variations
- Will standard icon sets be multicolor icons? Maybe only for light and dark.
- Will users be able to redefine a subset of icons to replace defaults?

> @mangelba I think you misunderstood my comment _What did you change?_ Clearly it is different but many of the icons appear the same. Unfortunately iirc your commit obfuscates the icon changes.

There are still no details of the changes.

flywire, 2024-07-20:
> I'm thinking we could create a script

Exactly: https://github.com/flywire/LibreCAD/commit/118474b64253008eaee066c79ae0ccda7eb4d913#commitcomment-129861123

> Converted in Windows 10:
> 
> `for %f in (*.svg) do sed -e 's/#000000/^\x0/;s/#ffffff/#000000/;s/^\x0/#ffffff/' %f`

zentus, 2024-07-20:
The problem is **Poor visibility of UI when using dark UI theme**. That could be solved via detecting whether the running OS is displaying the icons on a dark or a light background color.

Based on that we inverse the blacks and whites in the palette, at least for the icon outlines, and possibly changing some of the other icon colors to fit the white vs black icon outline style.

We could load both the light and dark theme images at build time and pick between those. Then we'd need a method to reload the whole icon set, or just let it reset on app reload.

Instead of loading icons like this

```c++
action->setIcon(QIcon(":/icons/select_intersected_entities.svg"));
```

We'd load them with something like this

```c++
std::string getIconPath(std::string filename) {
  //bool darkMode = isDarkMode();
  bool darkMode = true;
  std::string modeName = darkMode ? "dark" : "light";
  std::string iconPath = ":/icons/" + modeName + "/" + filename;
  return iconPath;
}

```c++
action->setIcon(QIcon(getIconPath("select_intersected_entities.svg")));
//=> ":/icons/dark/select_intersected_entities.svg"
```

zentus, 2024-07-20:
Most icons used via `setIcon` seem to be SVG, but some are PNG

[image]

zentus, 2024-07-20:
I ran the `build-all.yml` workflow in a fork. The `.dmg` is created, it mounts with the app in it, but the app does not run. Mac OS complains that the program has been "changed or damaged", and that the code does not match the signed code. Any ideas? Do I need some repo secrets or env variables?

dongxuli-concord-design, 2024-07-31:
Can you run our official DMG? 

If yes, could share the repository for me to test?

zentus, 2024-07-31:
> Can you run our official DMG?
> 
> If yes, could share the repository for me to test?

The official DMG works. [The one I built](https://github.com/zentus/LibreCAD/actions/runs/10018528215/job/27693984629) in [my fork](https://github.com/zentus/LibreCAD) does not work.

app4soft, 2024-08-23:
There is another theme concept: _Mid-Gray Color Theme stylesheet, compatible with Default Icons_.

- https://github.com/NabilAKROUD/LibrecadTheme

It looks good enough, just requires to tune active element green color tone, setting some missed UI elements colors from white to gray.

Also this theme has good compact icons style, so it is usable even on small 720p (non high-dpi) displays.

[image]

app4soft, 2024-08-23:
> 2\. Export ENV variable and run `librecad`
> 
> ```
> $ QT_QPA_PLATFORMTHEME=qt5ct librecad
> ```

Is not it missing `env`?

Also, tried with `qt6ct` for latest LibreCAD `2.2.2_alpha` AppImage (as it is Qt6 build) — it does not works (LibreCAD does not see `qt6ct` in available styles), maybe due to some AppImage related issues.

```Shell
$ env QT_QPA_PLATFORMTHEME=qt6ct ./LibreCAD*.AppImage
```

app4soft, 2024-08-25:
LibreCAD 2.2.2_alpha AppImage builds got the Dark splash now: https://github.com/LibreCAD/LibreCAD/releases/tag/2.2.2_alpha-latest

Its made poorly via workflow injecting two stings to change splash image and splash text color: https://github.com/LibreCAD/LibreCAD/blob/466d081db87d1e66d19ace9bc31ef52a0fc5a0e1/.github/workflows/build-all.yml#L41-L48
```Yaml
cp ${{github.workspace}}/desktop/media/splash/librecad01_dark.png ${{github.workspace}}/librecad/res/main/splash_librecad.png
sed -i 's/Qt::black/Qt::white/g' ${{github.workspace}}/librecad/src/main/main.cpp
```

If there are anybody famillar with macOS and Windows CLI/terminal analogs for described `cp` and `sed` commands, please, commit similar solution to `../workflows/build-all.yml` for Win and Mac respectively.

[image]

kepi, 2025-01-01:
If you just need Layer Tree readable, here is quick QSS. Just save it as `something.qss` and load it in `Options -> Widget Options -> Style Sheet`.

```css
/* Layer Tree */
QTreeView
{
    selection-background-color: #ccffcc;
    selection-color: #7CFC00;
}

QTreeView::item {
    background-color: #384247;
}
```

Also if you want quick icon fix suggested by @deric and you are on Arch, here is modified `librecad-git` aur PKGBUILD:

```sh
# Maintainer: Damien Flament <damien.flament at gmx dot com>
# Contributor: mickele <mimocciola at yahoo dot com>
# Contributor: Ilmari Repo <ilmari at gmail dot com> (librecad-svn PKGBUILD)
# Contributor: GazJ Gary James <garyjames82 at gmail  dot com> (CADuntu PKGBUILD)

_repository='LibreCAD'
pkgname=librecad-git
pkgver=2.2.2_alpha.latest.r225.g05bc6ccb7
pkgrel=1
pkgdesc="A 2D CAD drawing tool based on the community edition of QCad."
arch=('x86_64')
url="https://librecad.org"
license=('GPL')

depends=('muparser' 'qt6-svg' 'boost' 'freetype2')
makedepends=('git')
provides=('librecad')
conflicts=('librecad')

source=("git+https://github.com/LibreCAD/$_repository.git")
md5sums=('SKIP')

function pkgver
{
    cd "$_repository"

    git describe --long --tags | sed 's/\([^-]*-g\)/r\1/;s/-/./g'
}

function build
{
    cd "$_repository"
    cd librecad/res/icons
    find -name "*.svg" | while read i; do
            fname=$( basename "$i")
            fdir=$( dirname "$i")
            sed -i -e 's/#000000/#f2f2f2/g' "$i"
    done

    cd ../../../

    qmake6 -recursive librecad.pro \
        QMAKE_CPPFLAGS="$CPPFLAGS" \
        QMAKE_CFLAGS="$CFLAGS" \
        QMAKE_CXXFLAGS="$CXXFLAGS" \
        QMAKE_LDFLAGS="$LDFLAGS"

    make
}

function package
{
    cd "$_repository"

    install -m 644 -Dt "$pkgdir/usr/share/pixmaps" librecad/res/main/librecad.png
    install -m 644 -Dt "$pkgdir/usr/share/applications" desktop/librecad.desktop
    install -m 644 -Dt "$pkgdir/usr/share/man/man1" desktop/librecad.1

    cd unix

    install -Dt "$pkgdir/usr/bin" librecad ttf2lff

    install -Dd "$pkgdir/usr/share/librecad"
    cp --recursive -t "$pkgdir/usr/share/librecad" resources/*
}
```

u2fly, 2025-04-18:
Blender-like stylesheet, but need some tuning for LibreCAD (like layer tree fix https://github.com/LibreCAD/LibreCAD/issues/1144#issuecomment-2567007725)

- https://github.com/pgilfernandez/FreeCAD_Blender-stylesheet
  - requires this folder too: https://github.com/FreeCAD/FreeCAD/tree/main/src/Gui/Stylesheets/images_dark-light
  - (https://danielcregg.github.io/github-folder-downloader/)