LibreCAD/LibreCAD #1146: Copied text disappears when original zoomed/panned off screen.

tisik, 2019-09-26:
I'm using Ubuntu 18.04 and having a problem with copied text. The copied text disappears when the original disappears from the screen as the result of zooming in.

In the same way the problem exists when panning: when the original is panned off the screen, the copy isn't shown any more on the screen... When I zoom or pan the original text back on the screen, the copied text reappears.

### Operating System and LibreCAD version info
Version: 2.1.2
Compiler: GNU GCC 6.2.0
Compiled on: Sep 17 2016
Qt Version: 5.9.5
Boost Version: 1.62.0
System: Ubuntu 18.04.3 LTS

gerardbm, 2019-10-20:
Bug confirmed using the most updated version:

Version: 2.2.0-rc1-155-g10d502db
Compiler: GNU GCC 6.3.0
Compiled on: Oct 20 2019
Qt Version: 5.7.1
Boost Version: 1.62.0

SwimmingBird996, 2019-10-26:
I am having the same issue, sometimes rolling the scroll wheel a few times and returning to the same zoom level brings the text back into view.

Version: 2.1.2
Compiler: GNU GCC 6.2.0
Compiled on: Sep 17 2016
Qt Version: 5.9.5
Boost Version: 1.62.0
System: Ubuntu 18.04.3 LTS

PR #1181 (2020-05-29): Modifications: calculate borders of new entities
Fixes #810
Fixes #1146
closed 2020-05-29 by lordofbikes