LibreCAD/LibreCAD #1148: Dear LibreCAD is there a way to install your software without asking Admin rights?

racekiller, 2019-10-02:
### Expected behavior

### Observed behavior

### Steps to reproduce or sample file

### Operating System and LibreCAD version info

fa201, 2019-10-05:
On which system ?

racekiller, 2019-10-13:
Hi windows 10

gerardbm, 2019-10-20:
This is Operating System related, not LibreCAD related.

If you use Windows 10 as Admin user, not as regular user, it won't ask you for Admin permissions. However, why would you do that? It's dangerous and unsecure (giving free access of malware to your system32 and freely control your system). It's safer if you use it as regular user and giving it permissions when it asks you for them. In fact, if Windows were a safe OS, it shouldn't ask you a binary (Yes/No), it should ask you for the password (as in Unix-like systems).

flywire, 2019-11-29:
> It's dangerous and unsecure (giving free access of malware to your system32 and freely control your system).

I disagree. Anaconda python allows installation in user directory without admin rights. It is a very important feature in many corporate systems where users are locked out of system areas,

gerardbm, 2019-11-30:
The user directory != system32 directory.

flywire, 2019-11-30:
Agree that is the most secure install. So what? If corporate systems lock you out of that area it shouldn't stop you using the software.

lordofbikes, 2019-12-01:
follow up #695

git-tiger, 2020-03-04:
I am on a locked out system at work. Been using PortableApp version, although it is 2.13 only.

fa201, 2020-03-04:
@lordofbikes, if we find a way to that please open an issue on doc repo to remind us to show the necessary steps in documentation as this will definitely be useful for several Windows users. Thanks.

flywire, 2020-03-04:
Probably install it to `%USERPROFILE%\AppData\Local\Programs` and add to path statement.

flywire, 2020-03-05:
@lordofbikes Can you point me to creating the installer?

flywire, 2020-03-05:
@git-tiger A PortableApp might be a fairly trivial task (see https://portableapps.com/node/61720) but install without admin access would also be nice.

flywire, 2020-03-06:
Install requests (requires?) [admin access](https://github.com/LibreCAD/LibreCAD/blob/master/scripts/postprocess-windows/nsis-5.4.nsi#L44-L45). Try [NSIS Multi User Plugin](https://github.com/Drizin/NsisMultiUser#nsis-multi-user-plugin).

git-tiger, 2020-03-06:
If 2.2.0 is released, then PortableApps.com will release it.
Is there any reason why 2.20 is still in release candidate for so long?

flywire, 2020-03-06:
Please read the [PortableApps.com comment](https://portableapps.com/comment/242825#comment-242825) carefully.
- It implies if users consider v2.2.0-rc1 useful they would allow the PortableApp for that version built by someone else to be released on their site.
- **If LibreCAD Version 2.2.0-rc1 is useful users should advise them.**

If you [contact me](https://github.com/flywire) when I build the PortableApps version I'll send you a link so you can test and report on it.

git-tiger, 2020-03-06:
> We don't normally update to pre-releases, we only do stable releases. And it's based on what the publisher makes available.

This is what I read.
2.1.3 was released in 2016.
We are in 2020.

flywire, 2020-03-06:
Neither you nor I can control LibreCAD releases which are a more complicated process than you seem to understand and development effort is focused on LibreCAD_3. The PortableApps administrator asked _if LibreCAD Version 2.2.0-rc1 is useful users should advise them_ otherwise, as is normal, they _won't update [their PortableApps] to pre-releases_.

Would you mind testing [LibreCADPortable_2.2.0-rc1.paf.exe](https://drive.google.com/file/d/18BjTBqsVEMzKTbUn8EJD6rCHtDP7fNJR/view?usp=sharing) and giving us a User Acceptance Testing Report? It only needs to be a few comments about the environment(s) you tested it in and the results of the functionality tested. As a user of the current version you should know what to look for.

flywire, 2020-03-06:
Note #1198 request for installer silent install option

flywire, 2020-03-18:
Refer to discussion on [LibreCAD installer](http://forums.winamp.com/showthread.php?t=455633). Portable versions are configured differently but use the same install system.

flywire, 2020-03-26:
Trivial to change installer from HKEY_LOCAL_MACHINE (HKLM) to HKEY_CURRENT_USER (HKCU) and recompile. A few issues yet to resolve to allow either.