LibreCAD/LibreCAD #1149: Setting: Autozoom at Startup

cody-cshe, 2019-10-11:
Would it be possible to add an "Autozoom at Startup" checkbox in settings?
Most of the time when dxf's are first opened the item is not fully in view.

gerardbm, 2019-11-09:
Like this one?

[image]

Yes, this setting is in the last version of LibreCAD.

cody-cshe, 2019-11-09:
Yes, that looks good, which version is that, I can't find it in 2.2.0-rc1 or 2.1.3.

gerardbm, 2019-11-09:
I used the version of the master branch, by compiling the source code.

You can try to compile it as well from Windows.
https://wiki.librecad.org/index.php/Build_from_source

If you can't compile it, download and install the last binary (EXE) for Windows from the releases section, (this setting was probably already added in the version 2.2.0-rc1):
https://github.com/LibreCAD/LibreCAD/releases/tag/2.2.0-rc1

Keep in mind this:
- Last commit date: June 11, 2019 (if you compile the source code)
- Compiled file date: Feb 16, 2018 (if you download the binary EXE)

closed 2019-11-18 by cody-cshe