LibreCAD/LibreCAD #1152: Images disappear and don't change order.

javier4277, 2019-10-26:
### Expected behavior
 
1) Insert Image - move image 
 
2) Insert two Images - Change order
 
### Observed behavior

1) When I move an image, it is hidden and I have to do Auto Zoom so that it reappears,

2) If I add a second image, it does not work to change the order "move to hide or move to the front"

### Steps to reproduce or sample file

1) Insert Image - Move the Image 

2) Insert two images - Tools - Modify - Order -  Move to top or move to bottom

### Operating System and LibreCAD version info

MX linux 18.3 x64 Continuum

Libre cad Version: 2.1.2

dellus, 2019-12-19:
1. Fixed in 2.2.0 RC

2. First select the operation, for instance "move to top", then select the entities, close selection  with "Enter", then "Escape" twice. Same for drawing entities.

ncodernet, 2020-10-15:
On Windows with Version: 2.2.0-rc1images still disappear with several operations like move and scale.

dellus, 2020-10-15:
I can't reproduce this. Maybe you have an early 2.2.0 release. Get the
most recent version and check again.

ncodernet, 2020-10-16:
Sorry, you are right. I was not aware that there are so many builds of  2.2.0RC - I had used the one from https://github.com/LibreCAD/LibreCAD/releases - which is obviously very old. The June build from SourceForge works nicely .
Thanks...