LibreCAD/LibreCAD #1154: LibreCAD 2.2.0 RC1 crashing when moving simple square shape

80cm, 2019-10-27:
I am a new user to LibreCAD, and I have experience trivial crashes two times now. This will be about the crash I had just now, trying to use the snap tools, to move a box shape from A to B.

Computer:
Win 7 pro, 32GB ram

Expected behavior:
Software not crashing.

Observed behavior:
Software crashing. I was given a popup message about the software having crashed. Also no indication of what made it crash.

Reproducability:
Uknown

Version:
LibreCAD 2.2.0 RC 1

I am not able to account for everything I did up to the point of LibreCAD crashing, but I will try.
1) I started by loading a drawing I made the day before. A simple drawing of a wheel, maybe 20 different segments of lines of different shapes.
2) I am new to this software, so I wasn't sure how to use the snapping buttons, as they aren't intuitive to interpret, and sometimes mousing over them is time consuming, so I just tried one button at a time, for snapping a square shape and moving it around the round wheel I had drawn.
3) Not knowing how snapping or moving stuff really works, somehow, the software just crashed as I tried to use the snapping tools.
4) I have no idea what I did wrong. 
5) I did use the undo some moments before, but the crash doesn't seem directly related to an undo action.
6) I don't remember if I was maybe zooming in or out with mouse wheel immediately before it crashed.

Needless to say, as a user of GIMP, I have had basically zero crashes working with a really large pixel drawing, yet LibreCAD seem to crash too easily imho, for reasons unknown.

NOTE:  Isn't this weird, or am I doing something wrong?
1) Click move/copy button
2) Enable endpoint snap
3) Hover mouse over unselected box with endpoints
4) Nothing happens!
5) I expected the snapping to work on previously unselected objects, but that doesn't seem to work. Presumably related to why the software crashed, fiddling with both an object being either selected beforehand, and trying to snap to object with it being unselected.

80cm, 2019-10-27:
Another crash just now.
I move a square shape around a round wheel. This, so that I can scale the wheel down in size, to fit inside the square shape.

1) I did a window drag selection of both wheel and square shape, then unselected the square shape.
2) I clicked the scale tool. And with endpoint snap enabled, I clicked a point on the wheel. The popup menu then showed, and I entered .95 and delete original. 
3) Immediately after pressing OK, software crashed, giving me the popup, of the exe file having stopped working.

THEN

I did the very same thing as I did before, having started the software again, and loading a saved file.
This time, the software did not crash.

fa201, 2023-08-12:
Is this issue still relevant or did it improve ?

sand1024, 2026-09-26:
outdated

closed 2026-09-26 by sand1024