LibreCAD/LibreCAD #1156: "Save As" works only with modified drawings

Antonio-52, 2019-11-12:
### Expected: save current file with a different full filename, always, asking for the new name/path

### No effect at all: "Save as" works only when the drawing has been modified after last saving.

### Steps to reproduce or sample file
1. Open a file and modify it
2. Save the file.
3. "Save As" the same drawing. The file should be always "saved as", but no effect.
4. Modify again.
5. "Save As" the same drawing. The file is correctly "saved as".

### System: Ubuntu 18.04.3 LTS
### PC with 8 GB RAM
### LibreCAD: 2.2.0-rc1-242-gc4387f10
Compilatore: GNU GCC 5.4.0
Compilato in data: Aug 28 2019
Versione Qt: 5.5.1
Boost Version: 1.58.0

slackuser0xae34, 2019-12-11:
I've seen this too. The easy workaround is click the "Auto Zoom" button. This sets the "changed" flag and the "Save As" works.

It's nice to know that I'm not the only one who opens an existing drawing and saves it to a new name in order to create a new version/drawing from that existing drawing.

Antonio-52, 2019-12-15:
OK, I'll try.
But... this is simply a workaround, not very easy to discover.
It would be better if this will be corrected in the next (minor) version.
Thank you!
--
Antonio Angelo

Sent with [ProtonMail](https://protonmail.com) Secure Email.

‐‐‐‐‐‐‐ Original Message ‐‐‐‐‐‐‐

closed 2026-09-26 by sand1024