LibreCAD/LibreCAD #1157: Question: How to use this lib to just convert dwg to pdf in a Node.js app ?

Aarbel, 2019-11-21:
Hi guys,

Some of my users use .dwg and i'd like to convert them into .pdf to make them readable by [pdf.js](https://github.com/mozilla/pdf.js/) 
Can this library help me to do that ? My back-end is on Node.js.

Thanks a lot for your help & advices 👐

lordofbikes, 2019-11-22:
LibreCAD is an application, not a library.
[libdxfrw](https://github.com/LibreCAD/libdxfrw) is the library we use to read DWG and write DXF.
But the DWG support in libdxfrw is very rudimentary and I don't recommend to use it for DWG conversion yet.
Also I can't say, if it would even be possible to us a library written in C++ in Node.js, not my core competence.

closed 2019-11-22 by lordofbikes

Aarbel, 2019-11-25:
@lordofbikes you can do it with `execFile` method in node.js.

Ex with `ghostscript` :
[image]

sveisvei, 2020-04-01:
@Aarbel May I ask, I tested your  ghostscript command on a mac - and getting errors like `'Error: /undefined in AC1024\n`. Did you install anything in addition to make this work?

Aarbel, 2020-04-01:
@sveisvei i installed `ghostscript icc-profiles-free poppler-utils` 

Ghostscript is a piece of nightmare while debugging errors

sveisvei, 2020-04-01:
Aha, well - if nightmare is impossible! ;-).
Any tips on where to look for what the error might be: `Error: /undefined in AC1021...`

Aarbel, 2020-04-01:
@sveisvei be sure you have one of the last version of ghoscript installed. It depends of your debian version for example. 
Also verify your icc profiles are in the right path