LibreCAD/LibreCAD #1161: Crashes THE WHOLE SYSTEM when attempting to open a dwg

teo1978, 2019-11-23:
[wecancut.dwg.zip](https://github.com/LibreCAD/LibreCAD/files/3882681/wecancut.dwg.zip)

### Steps to reproduce or sample file

Try to open the attached file.

### Expected behavior

I have no idea whether there's something wrong with the file. If that was the case I would expect an error message. Otherwise, should open it.

I know it says "experimental, save your work first" but I assumed that meant your work within LibreCad.

### Observed behavior

LibreCAD hanged, i.e. stopped responding, and so did THE WHOLE OPERATING SYSTEM (Ubuntu).

The mouse cursor wouldn't move, the system wouldn't respond to clicks nor to the keyboard.

Not even Ctrl+Alt+F1 worked, so the only thing I could do was to force power down the computer by holding the power off switch. Outrageous.

Obviously, that the whole system stops responding because of the misbheavior of a single program (in this case LibreCAD), no matter how bad, is a bug in the OS which I have already reported. But clearly there's a severe bug in LibreCAD too.

No matter how experimental support for importing swg is, please at least try to catch exceptions or whatever and error out gracefully rather than hanging the user's computer completely.

### Operating System and LibreCAD version info

Ubuntu 16.04. Libre cad just installed few minutes ago via apt-get after adding the stable ppa. I'm not going to open this crap ever again, so don't expect me to go get the version information.

gerardbm, 2019-11-23:
I'm sure it's a Linux issue related on the RAM memory. Linux is very good managing the RAM memory, except when it ends without memory. Then the full system is frozen (Ubuntu, Debian, Arch and all the distros). When this happens, it's impossible to open a terminal to kill the process. You will have the same experience opening hundreds of browser tabs with Flash Player, for example.

By the way, there are several DWG versions, some of them are not even compatible on AutoCAD itself. Since DWG it's a propietary format, which may contain incompatible information, the recommended way of working is to use the DXF format.

I tried to open this file with the last version of LibreCAD and and the program crashed. This is the error:

```
QG_LayerWidget::activateLayer: invalid layer or nullptr layerView
QG_LayerWidget::activateLayer: invalid layer or nullptr layerView
terminate called after throwing an instance of 'std::bad_alloc'
  what():  std::bad_alloc
```

lordofbikes, 2020-02-07:
Follow up https://bugs.launchpad.net/ubuntu/+source/librecad/+bug/1853707

gerardbm, 2020-02-08:
Just for reference, about the Linux kernel bug: https://lkml.org/lkml/2019/8/4/15

So it would be fine if we could fix the LibreCAD side: the bug caused by the `libdxfrw` library.

bzuillsmith, 2020-08-15:
This happened to me in Windows 10 as well. Nothing actually hanged in my case, the app pretty instantly crashes and closes.

serval2412, 2021-01-29:
On pc Debian x86-64 with master sources updated today, I don't reproduce this.
Here are the only logs I got:
RS_DEBUG::setLevel(3)
RS_DEBUG: Critical
RS_DEBUG: Errors
RS_DEBUG: Warnings
libpng warning: iCCP: known incorrect sRGB profile
QLayout: Attempting to add QLayout "" to QG_LibraryWidget "Library", which already has a layout
QG_LayerWidget::update: nullptr lastLayer
QG_LayerWidget::activateLayer: invalid layer or nullptr layerView
QG_LayerWidget::activateLayer: invalid layer or nullptr layerView
RS_FilterDXFRW::endEntity(): updating hatch failed: invalid hatch area

lordofbikes, 2021-01-29:
This was indeed fixed recently in libdxfrw.
Many thanks for confirming that the fix works for this file.
I'm about to sort out more DWG issues, then I'll close it.

lordofbikes, 2021-01-31:
This was fixed with 973e81c9df803fcdd258f3c549bb664efd045bf9
Follow up to #988

closed 2021-01-31 by lordofbikes