LibreCAD/LibreCAD #1166: LibreCAD window unavailable, after multi monitor setup switched back to single monitor

lordofbikes, 2019-12-09:
### Expected behavior
On a notebook with external monitor LibreCAD window is moved to the 2nd screen.
When the notebook is started without external screen, the LibreCAD window should appear on the main screen again.

### Observed behavior
LibreCAD window is unavailable. The icon is in the taskbar, but the window is out of the screen.

### Steps to reproduce or sample file
- setup multi monitor environment
- start LibreCAD and move the window to the second screen
- close LibreCAD
- remove extra monitor
- start LibreCAD again

### Operating System and LibreCAD version info
Windows 10
LibreCAD 2.2.0-rc1

Reported in forum: https://forum.librecad.org/Main-window-not-visible-tp5718099.html

lordofbikes, 2019-12-09:
The reported work around for Windows is to minimize all windows (shortcut WIN-m) and restore LibreCAD by clicking the task bar icon.

Another Windows work around is to edit LibreCAD settings in registry (while LibreCAD is closed):
HKEY_CURRENT_USER\Software\LibreCAD\LibreCAD\Geometry
Check WindowHeight, WindowWidth, WindowX and WindowsY and set appropriate values.

On Linux and MacOS the settings are stored in LibreCAD.conf somewhere in the user home folder.

Sargas0X0101, 2021-01-08:
Hello all
i have the same problem.
I work with a Laptop in a docking station with 2 external monitors.
If i open LibreCAD i have the Icon in the Taskbar but no window.
The workaround with win+M and taskbar icon does not work.

i have a Win 10 64 bit system

my regedit values are : 
[image]

what are the correct values for window x and y ? 

Thanks

lordofbikes, 2021-01-08:
There are no correct values for WindowX and WindowY, they simply contain the last window position.
Your values represent negative numbers, means your LibreCAD window was left side of the main display.
So setting both to 0 should bring back LibreCAD to the main monitor again, no matter where the unplugged display was located.

kn0xit, 2022-12-27:
I have the same problem running LibreCAD 2.2.0 on Windows 11.

> The workaround with win+M and taskbar icon does not work.
Same here.

I had to uninstall, delete LibreCAD folder in Windows registry, and reinstall to get it working again.

fa201, 2022-12-29:
This is only applicable to Windows OS ?

kn0xit, 2022-12-29:
In my case, yes, as the resolution of the issue is also tied to the Windows Registry entries.

bajraan, 2022-12-30:
There is a simple way how to bring the window back. 
1. Click on the libreCAD on the taskbar
2. Alt + Space (Small window appear in the upper left corner)
3. Click on Move or press M
4. Using arrows you can easily bring back the LibreCAD window.

[image]

dxli, 2023-07-02:
Commit a6f3423

Tavo965, 2023-11-12:
This is what worked for me:
1-	Close LibreCAD
2-	Go into the screen configuration of windows
3-	Change de order, if screen 2 is on the right, move it to the left
4-	Launch LibreCAD, the main screen should be there