LibreCAD/LibreCAD #1173: Print Preview - icon missing for tiled printing

motobros-calgary, 2019-12-14:
### Expected behavior  - tool bar icon

### Observed behavior - icon is missing

### Steps to reproduce or sample file  n/a

### Operating System and LibreCAD version info  - v2.2.0

I wanted to try using "tiled printing" for a project. According to the user manual there should be a icon on the tool bar for this purpose but I have not been able to find it after looking everywhere.

Anyone else have this difficulty or know of a solution ???

Thanks

baranovskiykonstantin, 2019-12-14:
Have you built LibreCAD from sources?
This feature is available in master since [863a2cb](https://github.com/LibreCAD/LibreCAD/commit/863a2cbf9311871ad515c085d030baa19060d82d).

motobros-calgary, 2019-12-15:
Hi, thanks for responding.  I have done very little coding on anything, not really a big interest for me.  However,  I have really enjoyed using Librecad for some designing and just recently did a design where having Tiled Printing would be very useful.   Is there a version of Librecad available with the tiled printing feature alread built in ??

Bye for now,

Dave Fisher

⁣Get BlueMail for Android ​

baranovskiykonstantin, 2019-12-16:
If you are using Windows or OSX you can get the latest installer with tiled printing support on [sourceforge](https://sourceforge.net/projects/librecad/files/).
If you are using Linux you should search the properly version in your repositories.

fa201, 2019-12-17:
You can build LibreCAD from source using this page: https://librecad.readthedocs.io/en/latest/appx/build.html
Hope it will help.

fa201, 2023-08-12:
[image]

This is solved on 
Version: 2.2.1_alpha-274-g45a167b0
Compiler: GNU GCC 11.4.0
Compiled on: Aug 12 2023
Qt Version: 5.15.3
Boost Version: 1.74.0
System: Ubuntu 20.04.6 LTS

sand1024, 2026-09-26:
tiled printing is supported

closed 2026-09-26 by sand1024