LibreCAD/LibreCAD #1189: It's not possibile to fillet 2x a corner of a rectangle

tormec, 2020-01-26:
### Steps to reproduce the issue
1. create a rectangle
2. fillet an angle with a generic radius
3. repeat the fillet on the same angle with a bigger radius
4. the angle is not rounded off (while working on an angle between two lines the fillet command work as expected) 

### Operating System and LibreCAD version info
Version: 2.1.2
Compiler: GNU GCC 6.2.0
Compiled on: Sep 17 2016
Qt Version: 5.9.5
Boost Version: 1.62.0
System: Ubuntu 18.04.3 LTS

lordofbikes, 2020-01-26:
This seems to be related with polylines, a rectangle is created as polyline.
A work around is to explode the polyline, then fillet works multiple times on the same lines.

Presumably this happens because the two lines are separated by the fillet arc. They have no direct connection after the first fillet.

Murmele, 2021-07-06:
I think this is similar for creating a fillet between a Spline and a two point line or am I wrong?