LibreCAD/LibreCAD #1192: Trim by two limiting entities

MikePryadko, 2020-02-03:
[image]
I have circle and two lines: I would like to remove {red arrows} and keep {green arrows} - at the moment it's impossible.

### Operating System and LibreCAD version info
Version: 2.1.2
Compiler: GNU GCC 6.2.0
Compiled on: Sep 17 2016
Qt Version: 5.9.5
Boost Version: 1.62.0
System: Ubuntu (xubuntu) 18.04.4 LTS (kernel: 5.3.0-28-generic x86_64)

lordofbikes, 2020-02-03:
Trimming circles is a bit tricky. For me these steps work:
[image]
* select 1st line, click at 1
* trim circle, click at 2

result is a semi circle
* select 2nd line, click at 3
* trim circle, click at 4

result is what you are looking for

I assume behavior depends on the circle direction and line angles.
Clicking 3/4 first and then 1/2 works too, but it's not so obvious because the result between 3/4 and 1/2 is a 360° arc and not a circle anymore. This difference is not visible, it seems that nothing has happened, but it has.

To avoid this issue you can use the divide tool. Use intersection snap and divide the circle at the line intersections. Then the unneeded part could be simply deleted.

Anyhow, this must be investigated and the workflow should be the same, independent which line is selected first. Many thanks for reporting.

fa201, 2020-02-03:
Same conclusion but I found it very difficult to trim the circle (10 attempts to achieve it).
I could not trim a rectangle either (to create a chamfer for example).

Version: 2.2.0-rc1-155-g10d502db
Compiler: GNU GCC 7.4.0
Compiled on: Jun 15 2019
Qt Version: 5.9.5
Boost Version: 1.65.1
System: Ubuntu 18.04.4 LTS

MikePryadko, 2020-02-04:
@lordofbikes oh, `trim` works in both directions (truncate & lengthen)... unexpectedly
(from google: _"To trim - make (something) neat or of the required size or form by cutting away irregular or unwanted parts"_)
I've tried to use `lengthen` (from google: _"To lengthen - make or become longer"_) for that purpose, but it makes strange things :neutral_face: (I have no idea when that tool can be useful)

__My suggestions__:
1. `trim` only truncates lines and allow to select TWO limiting entities
1.2 if user needs only one - he just press Enter to skip the second
1.3 do not allow user to trim circle when it has only one intersection with the limiter (now LibreCAD is extending the limit line - I find it completely useless)
2. `lengthen` makes the lines longer up to the limiter or by the specified value (it would be perfect to have a choice: add to line/circle some part or make it exactly specified length)

jimls, 2020-02-07:
When I have a rectangle and a line crossing it I cannot select the rectangle as the limiting item (to trim the line).  I have to put a line in temporarily where the side of the rectangle is and trim to that, then delete the short line segment.  Seems to be a bug but perhaps there is a better way to do it.

lordofbikes, 2020-02-07:
A rectangle is created as polyline which does not support trimming by design. To use the rectangle for trimming or trim it itself, you must explode it first.

MikePryadko, 2020-02-08:
> ...which does not support trimming by design ...

You see, it would be useful to have an ability to use multiple objects as "limiting entity" :stuck_out_tongue:
In case of polyline it's just an array of lines so when use select it you just got selected all of them - and work with it like with usual lines :smirk:

lordofbikes, 2020-02-08:
Oh, when it is just so simple, I'll start right now :wink: 
But seriously, when I say *by design*, I mean it's not implemented yet. And I agree, that this is a missing feature.
It's not that easy to implement if someone takes the challenge.
* polylines are a mix of lines **and** arcs
* selecting a whole polyline can bring lots of intersections, how to choose which are the right ones
* selecting single entities from polyline is not supported by basic internal functionality

That's just what came to my mind yet, there may be more issues to solve to implement this.

Using more than one entity for trimming is an often asked request too. This is not related to polylines, but can make it easier when implemented first.

nepomnyi, 2020-10-08:
I also do need trimming by two limiting entities very very much!

I guess the difficulty with making it is associated with how your trim feature is implemented. Thus, when trimming I am required to select the portion of the object that needs to be left, whereas in AutoCAD I am required to select the portion of the object that needs to be cut off.
So, to implement it you might need to rewrite your entire code for trimming ...
May be one day, I will be able to contribute to it ...

Meanwhile, thank you so much for LibreCAD!
I can only hope you'll make this issue your top priority because that's one of the very common things and, therefore, limits one's work with LibreCAD.

Thank you very much again!
Ivan

jcbeppler, 2021-07-12:
@nepomnyi I was wondering the same thing that if the process is made in the way that AutoCAD is done, like clicking in the place to be trimmed, the logic may be easier to implement.
I took a short look over the code and probably this need to be changed for every drawing tool, it may be a quite big change.