LibreCAD/LibreCAD #1193: Copy/paste into a file fails if current layer is locked

fa201, 2020-02-13:
### Expected behavior
Paste a drawing into another one (in the same session) while the current layer in destination drawing is locked.

### Observed behavior
Paste seems to be done but the pasted parts disappear after a blink. Refresh and toggling off then on all the layers does not show the parts. Yet the source layers are paste.
Unlocking the current layer makes the paste working.

### Operating System and LibreCAD version info
Version: 2.2.0-rc1-155-g10d502db
Compiler: GNU GCC 7.4.0
Compiled on: Jun 15 2019
Qt Version: 5.9.5
Boost Version: 1.65.1
System: Ubuntu 18.04.4 LTS

TomaszPluta, 2020-03-18:
> Yet the source layers are paste.
Could you elaborate this?

Whycshould one expect to paste into a locked layer?

fa201, 2020-03-18:
I did not mean to paste into a locked layer. I meant pasting a layer from drawing A into drawing B while drawing B current layer is locked. The behavior seems not right. I think that either LC should prevent from doing this, either it should allow it and do it. That's the purpose of the issue.

TomaszPluta, 2020-03-19:
How can a layer be copied? I tried to reproduce, but i can  only select items and paste selected intems (there is no option "copy layer" in layers contex menu).

fa201, 2020-03-20:
Sorry, I meant copying layer's entities.