LibreCAD/LibreCAD #1194: Export as PDF produces a tiny or a huge blank pdf

liesauer, 2020-02-22:
### Expected behavior
convert dxf to pdf successfully

### Observed behavior
a tiny or a huge blank pdf
[image]

### Steps to reproduce or sample file
1. open my dxf
2. export as pdf

### Operating System and LibreCAD version info
osx 10.13, LibreCAD: 2.1.3

try testing with this dxf
[test.dxf.zip](https://github.com/LibreCAD/LibreCAD/files/4238642/test.dxf.zip)

fa201, 2020-02-22:
The drawing is on layer 0 which is normaly not a good practice; Did you try to create a new layer and move entities to this new layer ?

dellus, 2020-02-23:
In Options - Current Drawing Preferences - Paper set a valid papersize first, for example A4.
Then go into Print Preview and leave it again. You don't have to make any changes there, it just must have been running.
Now Export to PDF will get a result.
This issue has a pending pull request. The fix has not been incorporated yet.

liesauer, 2020-02-25:
> The drawing is on layer 0 which is normaly not a good practice; Did you try to create a new layer and move entities to this new layer ?

yes, i tried creating a new player but still blank pdf. but @dellus 's advise works!

liesauer, 2020-02-25:
@dellus it works! thanks!

closed 2020-02-27 by liesauer