LibreCAD/LibreCAD #1199: Adding any Dimension freezes Application after selecting the two reference points.

mkloock, 2020-03-02:
### Add the dimension and be able to position it in the third click

### Application freezes with spinning wheel after selecting the two reference points.

### Draw rectangle or line, add a dimension (choose align for example). 
* click on two endpoints
* observe freeze

### Mac OS 10.15.3 (Catalina) System and LibreCAD 2.1.3

lordofbikes, 2020-03-02:
This is an often used feature and works for many users, so possibly this can be related with drawing preferences.
Please make a test case drawing.
* create the rectangle
* save the drawing
* check that the application freeze by drawing the dimension
* upload the saved file here for further investigation

mkloock, 2020-03-02:
OK - I was surprised, too that this common functionality would crash consistently.  Me being a new user, I now found an oddity that was causing this:

When I installed first time, I set the unit to Microinch. That apparently set a lot of other settings for dimenions to really odd settings such as text height over 9000. The combination of these settings made the application crash when applying a dimension. Now that I set everything back to 1 and I am using inch, the problem went away.

This is probably still a defect but a very rare one I would say.

closed 2020-03-02 by mkloock

lordofbikes, 2020-03-03:
Many thanks for the feedback.