LibreCAD/LibreCAD #1202: Crash when saving a block

Jim-2249, 2020-03-07:
### Expected behavior
Would save block just as any other

### Observed behavior
Crashes while saving block

### Steps to reproduce or sample file
1) open LibreCAD
2) open Crank circuit.dxf
3) select Block List tab
4) select Bridge three phase
5) click on "save the active block to a file"
6) wheel appears then LibreCAD closes

### Operating System and LibreCAD version info
Windows 10 pro 64
Version: 2.2.0-rc1-142-ge38b7088
Compiler: GNU GCC 4.9.1
Compiled on: May 17 2019
Qt Version: 5.4.1
Boost Version: 1.53.0
System: Windows 

I don't seem to be able to upload a .zip file, here is a OneDrive link
https://1drv.ms/u/s!Ajzp9H2J5N1LiEk5JIbbgOcjLdcr?e=bof8Hj

TomaszPluta, 2020-03-09:
I can consider, reproducible. It ends with:
_malloc_consolidate(): invalid chunk size_
I will take a look at it.

baranovskiykonstantin, 2020-03-15:
This issue has a fix: #1180.

commit 39300516 (2020-05-22): Merge pull request #1180 from baranovskiykonstantin/saveblockfault, fixes #1202

Fix saving to file of the block with inserts.
closed 2020-05-22 by lordofbikes