LibreCAD/LibreCAD #1205: Slow saving over network

git-tiger, 2020-03-18:
### Expected behavior
Saving and editing files should be the same as on local drive, instantaneous.

### Observed behavior
Saving an edited dxf file on the network drive is very slow.
Saves at rate of around 1kB/s.

### Operating System and LibreCAD version info
Windows 10
2.2.0rc1 portable

lordofbikes, 2021-06-26:
cross links:
https://forum.librecad.org/LibreCAD-Slow-Saving-Across-a-Network-tp5714702p5720797.html
https://serverfault.com/questions/808633/small-writes-to-smb-network-share-are-slow-on-windows-fast-over-cifs-linux-moun

GSStnb, 2021-06-27:
I poked around libdxfrw.cpp/.h which lead me to /intern/dxfwriter.cpp.  Line 19 of  /intern/dxfwriter.cpp contains a comment "//RLZ TODO change std::endl to x0D x0A (13 10)".  Seems RLZ knew it might be a problem.  "endl" is used in several times in write methods  (e.g. writeString [line 221], writeInt16 [line 232], ...)

Characters "13" and "10" are ascii 'cr' and 'nl'.  Would it be enough to replace "endl' in those write statements (e.g. line 221, 232, ...)?  Are both 'cr' and 'nl' (carriage return and newline) required, or just the newline (\n)?

I'll try to set up Win10 VM to make the changes and test it...

lordofbikes, 2021-06-27:
Using `"\n"` is enough. It's the compilers responsibility to translate it to the native OS line ending format.
Replacing `std::endl` with `"\n"` should basically work, but I consider to use a `dxfWriter::endl()` method, which would open some opportunities for future use.

GSStnb, 2021-07-19:
I set up a Win10 vm and changed the "endl" to "\n" for the dxf write operations (dxfWriterAscii::writeString, etc).  It made no difference in saving to a network share - still very slow.  Saving to a local drive remains quick.

erictmyers, 2022-06-27:
Hi, can a new installer be built including this fix, and added to the nightly builds?

lordofbikes, 2022-07-02:
The above `libdxfrw` contribution LibreCAD/libdxfrw#62 was merged and also pulled into LibreCAD.
But from forum discussions I can't say if this fix will solve this Windows share issue too.
The installer containing this fix will be named `2.2.0-rc4-8-g5a07ab41`.
It is available here soon and tomorrow on Sourceforge too.

erictmyers, 2022-07-06:
Thanks, I can confirm that did not fix the issue.