LibreCAD/LibreCAD #1207: Exiting print preview displays wrong drawing

xanderdin, 2020-03-18:
### Expected behavior
Exiting print preview mode should return to previewed drawing document.

### Observed behavior
Exiting print preview mode brings to another drawing document.

### Steps to reproduce or sample file
1. Run LibreCAD (default 'unnamed document 1' will be opened).
2. Open another document with some drawing.
3. Click Print Preview icon in toolbar to enter preview mode.
4. Click Print Preview icon again to exit preview mode.

Result: instead of returning to the opened in step 2 document the empty 'unnamed document 1' is displayed.

### Operating System and LibreCAD version info
Version: 2.2.0-rc1-190-g51b21cc7
Compiler: GNU GCC 8.3.0
Compiled on: Mar 18 2020
Qt Version: 5.11.3
Boost Version: 1.67.0
System: Debian GNU/Linux 10 (buster)

I've also build @baranovskiykonstantin's version from https://github.com/baranovskiykonstantin/LibreCAD/tree/closeprintpreview for comparison and it does not have this issue. Obviously the rejected https://github.com/LibreCAD/LibreCAD/commit/2bd7b074ecced9545fa99940ab20d13b44567472 is somehow responsible for this.

lordofbikes, 2020-05-21:
I've merged the rejected commit, but the issue remains for me.
Also, the same issue applies when a block is edited and closed.
- start a new session in tab mode, which opens an unnamed document
- open a drawing with blocks
- edit any block
- close the block editor tab
- the first tab, which is the unnamed document is activated

I assume there is no and never was any logic behind the tab handling. So when any tab is closed, the index for the active tab is reset to 0 which simply is the first tab.
This would need a kind of stack to implement, which could also be used for CTRL-Tab and CTRL-SHIFT-Tab handling.
Currently these hot keys iterate through the tabs ascending or descending. With some logic behind we can implement the known functionality from other software, where CTRL-(SHIFT)-Tab iterates the tabs in historical order. This allows to fast toggle between two drawings over and over again.

xanderdin, 2020-05-21:
I confirm, the issue is still there.

Strange, but it looks like I've mistaken when was comparing builds and built not from baranovskiykonstantin:closeprintpreview  but rather from his master branch which was not having the issue. :man_facepalming:

tonischranz, 2021-05-09:
Hi there, today i experienced this behavior too, on 2.2-RC2 freshly built on FreeBSD... is anyone currently working on that? Or may I offer to give it a try? (btw: in the About dialog it says 2.2.0-alpha, with that line from make install "install -s -m 555 /usr/ports/cad/librecad/work/LibreCAD-2.2.0-rc2/unix/librecad", because of that i originally removed the pkg and built it from ports)

fa201, 2021-05-09:
I have the same on Ubuntu. 
Version: 2.2.0-rc2-15-g76828565
Compiler: GNU GCC 9.3.0
Compiled on: Apr 30 2021
Qt Version: 5.12.8
Boost Version: 1.71.0
System: Ubuntu 20.04.2 LTS

The first tab is also active when I close the tab for the opened DXF file.

@tonischranz, you can give it a try. Thanks.

tonischranz, 2021-05-10:
> @tonischranz, you can give it a try. Thanks.

Ok, i have a look at the port to figure out how to build the repo... seemed to need cmake and probably qmake first.

tonischranz, 2021-06-01:
Ok, as far as i could see from a quick glimpse (not yet really dived deep into it)  what is probably missing is a kind of stack that reminds which document was opened last, since the method called after closing one view is something like getLast or getCurrent or so on the views-collection. As far as i can remember, i somehow took part several years ago on a discussion of a very similar topic and i think the consent at that time was to implement a stack based on a linked list, where you basically just reorder the views (last opened to top or bottom) to avoid the stack growing endlessly.... well so i think the point to attack would be the opening of views rather than the closing. Nevertheless i'm not sure if the collection that is used for views is either Qt stuff or somehow used for other stuff, would have to check that first, and find an appropriate solution. So long

closed 2023-10-07 by lordofbikes

intrixius, 2024-02-06:
v2.2.0.2 has similar bug; when exiting print preview a new "unnamed document 1" is opened instead of the current document

commit 4cfa8d9f (2023-10-07): fixes #1207, Exiting print preview displays wrong drawing (#1696)

* fix: Exiting print preview displays wrong drawing #1207
commit 6f498163 (2023-10-07): fixes #1207, Exiting print preview displays wrong drawing (#1696)

* fix: Exiting print preview displays wrong drawing #1207