LibreCAD/LibreCAD #1210: Specify length of next line

brendanjodowd, 2020-04-05:
### Feature request

In line mode, after you've entered the first point, it would be great if you could specify the next point by entering the length, with the line then being produced in the direction of the cursor. This is especially handy in ortho mode for quickly drawing up plans, and this is the main mode of drawing for me in AutoCAD. So to draw a line to the left of an existing point, rather than entering "@-100,0" , you would move the cursor to the left and type "100" (enter).

susnicek, 2020-06-11:
Hi, in fact this is somehow possible this way:
1) enter your first point
2) write down your distance e.g. "100" (in the command line)
3) then hold "ctrl" key and press an arrow key (left, right, up, or down). According the arrow direction you will draw a vertical/horizontal line with the given length.

Best regards, Susnicek

DuncanLithgow, 2021-04-03:
@susnicek is that method documented anywhere?

susnicek, 2021-04-14:
Hi, I don't think that the method is documented because it works for me only in development version of Librecad or in rcx 2.2.0 versions. 
I noticed the possibility in the code...

DuncanLithgow, 2021-04-16:
This is in the documentation for the coming 2.2 release: https://librecad-docs-dev.readthedocs.io/en/latest/ref/fundamentals.html#drawing-and-editing-entities

natrius, 2021-04-23:
@DuncanLithgow 

This should be opened because this is not what was requested. What @susnicek wrote is a good function but is not what @brendanjodowd requested, because you can just draw with 90° angles. I try to explain what he means:

Press `l`, set with the left mousebutton the first point, drag the mouse in one direction, enter `100`, press enter, line is drawn in the direction with the entered length.

# Example:

[image]

I know the distance between the two house-corners. So i set the first point on the top left, drag my mouse where you see it on the screenshot, enter the distance (for example 34 meters) and press enter and the line gets drawn with the desired length.

This was also mentioned in the forum by someone coming from AutoCAD i think, but i cannot find the thread right now. I think someone mentioned there, that not every libreCAD user used AutoCAD before, hinting to the lack of explanation.

brendanjodowd, 2021-04-23:
@DuncanLithgow 
Yes that's it exactly. The way you've described is how i would operate in AutoCAD 90% of the time, specifying distances in ORTHO mode.

DuncanLithgow, 2021-04-23:
Okay, let's open the discussion. It was narrow request with answer... but let's look wider.

AutoCAD does really have some great ways to draw from the keyboard. For example, with cursor at 100,50 I can write a new line back to 0,0 either absolute with line > 0,0 or relative with line > @-100,-50. There's also a way from the keyboard of defining angle and distance.

GSStnb, 2021-04-23:
@DuncanLithgow 

> AutoCAD does really have some great ways to draw from the keyboard. For example, with cursor at 100,50 I can write a new line back to 0,0 either absolute with line > 0,0 or relative with line > @-100,-50. There's also a way from the keyboard of defining angle and distance.

LibreCAD does the same thing.  Coordinates can be written 10, 50 (Absolute Cartesian ), @10,50 (Relative Cartesian), 10<45 (Absolute Polar) or @10<45 (Relative Polar)

See: https://librecad.readthedocs.io/en/latest/ref/fundamentals.html#

natrius, 2021-04-23:
Thanks for opening again. 

Yupp, agree, thats all working. Missing is the "point mouse (with or without ortho fang 15° steps?), write `100` (without `@`, without `,`, without an addtional number) enter, line finished" approach.

GSStnb, 2021-04-23:
@natrius 

> [...] Missing is the "point mouse (with or without ortho fang 15° steps?), write `100` (without `@`, without `,`, without an addtional number) enter, line finished" approach.

Not precisely what you are looking for, but holding the [Shift] key while drawing with a mouse will provide constraints at 15° (be sure all the snaps are off).  See: https://librecad-docs-dev.readthedocs.io/en/latest/ref/fundamentals.html#drawing-and-editing-entities

natrius, 2021-04-24:
@GSStnb 

Well, yes, thanks, but thats not the point of this issue :)

1. enter `l` to draw a line
2. press `enter`
3. left mousebutton to start position (start drawing a line)
4. drag mouse in the direction where you want to draw
**5. Enter `100` (without `@` or any other symbol or additional number)**
6. press `enter`
7. Line is drawn, 100 in the direction your mouse was pointed when pressing enter.

Point 5 is not working or not possible. But its a good way to draw often some things way faster. Drag the mouse, enter number, enter, done. Next line. And so on. Its something quite a lot AutoCAD users are used to because its used often, as brendan mentioned above already.

Yes, i can use `SHIFT` while dragging to draw a line but i don't know how long the line will be. Its a good guess at best. The ortho is something for another issue. :)

GSStnb, 2021-04-24:
@natrius 

> Yes, i can use `SHIFT` while dragging to draw a line but i don't know how long the line will be. Its a good guess at best. The ortho is something for another issue. :)

Okay, I get that, but just thinking out loud - in the same way you don't know how long the line is going to be, how do you ensure the correct angle?  Just by the direction of the mouse movement? Or is this a combination of a specified line length and a constrained angle (e.g. ortho or 15° steps).

natrius, 2021-04-24:
I knew that question was coming and you are correct :D I had already a follow up issue prepared already actually for a ortho with 15° steps and more but deleted it for now because i thought it would not make much sense until this is implemented. :)

For now, just imagine you want to draw from a imported image like i did above. Its an pdf/image footprint of an castle i want to bring into CAD for measuring and so on.  It is just an example for now. 

But when i was using AutoCAD more actively, i was using this pretty regular because it makes it so easy to draw soemthing in combination with the mouse.

brendanjodowd, 2021-04-24:
@natrius Your description is spot on. Ideally the line command would still be active after point 7. I have never really required 15° intervals in autocad, if i wanted angles in conjunction with this mode of drawing I would adjust the UCS.

brendanjodowd, 2021-04-26:
From memory, I think AutoCAD had a related feature, which would be nice to incorporate with the type of drawing discussed already. It comes in to play at 3 and 4(b) below. 
1. type "l" (for line), press enter.
2. left-click anywhere for first point
3. type "<30", press enter - next point is now restricted to a line at 30 degrees to the horizontal going through the first point
4. Two options for second point:
4(a). left-click for second point, which appears at the closest location to the cursor on that 30 degree line.
4(b). (of relevance here) move the cursor towards the preferred direction of the line (30 degrees [top right] or 150 degrees [bottom left]), then type "100", press enter. You have created a line of length 100 which is at 30 degrees to the x-axis and which goes through your first point. 

Someone with access to AutoCAD might verify 4(b) for me. Anyway I think it would be a convenient, intuitive tool.

DuncanLithgow, 2021-05-01:
I think it would be good if we stayed on topic. Other similar features might be great - but one topic at a time. Discussion is for the forum.

For this specific example I just checked how QCAD does it.

1. "LI" (for line)
2. SPACE (puts cursor in cli)
3. "0,0" (puts first point at absolute 0,0)
4. Move the mouse somewhere and write the distance, for example "100". The second point is now 100 units in the mouse direction.

natrius, 2021-05-01:
Don't know how you did it, but i was not able to replicate it. I'm getting a "Unknown command: 100" when entering `100` and pressing enter.

[image]

lordofbikes, 2021-05-02:
> For this specific example I just checked how <h2>**QCAD**</h2> does it.

brendanjodowd, 2021-05-04:
@DuncanLithgow I think your summary is pretty good. Not sure if this goes without saying but this should also work with the polyline function.

natrius, 2022-01-19:
@DuncanLithgow 
I think the "implemented" tag is wrong in this case? Any updates by chance? :)

alandema, 2022-04-14:
To heve this feature would be great.

sand1024, 2026-09-26:
supported by snake command, and relative input assistant popup

closed 2026-09-26 by sand1024

natrius, 2026-09-26:
@sand1024 I don't think this issue is solved. I downloaded 2.2.1.5 on Windows 10. I found then snake command  in the menu 

[image]

and i tried it, but i struggle to find the difference to some lines drawn with `snake` or `line`. Also, it is not possible to specify a distance. I then used "snake (x)" and found the "specify distance" there. 

Thats a good starting point, but still a far point from what is desired in this proposal: 

1. enter l to draw a line
2. press enter
3. left mouse button to start position (start drawing a line)
4. drag mouse in the direction where you want to draw (possible restrict to ortho when pressing shift, like 7,5° or 15° or 30° or 90°(!) )
5. Enter `100` (without @ or any other symbol or additional number)
6. press enter
7. Line is drawn, 100 in the direction your mouse was pointed when pressing enter.

So, basically should be "either drag the mouse and leftclick for the next point, OR specify distance" like in the "snake (x)" or "snake (y)" command, but just in the direction the line is currently showing. Maybe restricted with shift.