LibreCAD/LibreCAD #1212: 2.2.0 RC build for MacOS is missing

lagivan, 2020-04-08:
It would be great to have MacOS build (dmg) of 2.2.0 RC available in Releases section.

jsissom, 2020-06-07:
It can't be built.  I tried to build using clang from the 2.2.0 rc tag commit 58deefa34cb07a3b7b4ed97cd1ac70f9b7fc56d0 and got many errors like "error: incomplete type 'QAction'".

lordofbikes, 2020-06-08:
@lagivan please find them at https://sourceforge.net/projects/librecad/files/OSX/2.2/preRelease/
Not automated yet, but maybe soon.

lordofbikes, 2020-06-08:
@jsissom you have to share more detailed information if you expect any help.
MacOS, Clang. Qt versions?
What have you done so far?
Qt Creator or command line?
Why this specific commit from 2018 and not the latest master?

jsissom, 2020-06-08:
I don't expect any help thank you.  No one had replied to lagivan for a month so I replied letting them know the code for the 2.2.0rc tag wouldn't compile on the mac using the instructions provided in the project.  I tried HEAD on master and it works so it has been fixed.

bettabacon, 2020-08-19:
@lordofbikes Since @jsissom got it working, is it possible to get a DMG up on releases?

> I don't expect any help thank you. No one had replied to lagivan for a month so I replied letting them know the code for the 2.2.0rc tag wouldn't compile on the mac using the instructions provided in the project. I tried HEAD on master and it works so it has been fixed.

DuncanLithgow, 2021-04-03:
I'm no expert here. Please tell me what is needed and what is solved to help Mac users get better access to LibreCAD. 

@jsissom do you have a compiled version we can share? Is the documentation wrong on how to install for Mac?

jsissom, 2021-04-04:
I don't have anything to share.  At the time, I followed the instructions in the project against HEAD and was able to compile it but the 2.2.0 RC version did not compile.  I moved on to qcad so I no longer use this project and can't be of help.

dxli, 2024-05-18:
Closing for 2.2.1_rc and 2.2.2_alpha

closed 2024-05-18 by dxli