LibreCAD/LibreCAD #1215: Help: commands linkshould be updated.

fa201, 2020-04-14:
### Expected behavior
Menu **Help/On line/Commands** should redirect to https://librecad.readthedocs.io/en/latest/ref/tools.html

### Observed behavior
Menu **Help/On line/Commands** currently directs to deprecated Wiki.

### Steps to reproduce or sample file

### Operating System and LibreCAD version info
Version: 2.2.0-rc1-190-g51b21cc7
Compiler: GNU GCC 7.5.0
Compiled on: Mar 29 2020
Qt Version: 5.9.5
Boost Version: 1.65.1
System: Ubuntu 18.04.4 LTS

lordofbikes, 2020-12-29:
The link to the wiki is currently the best place with the command line commands, which is the intention behind this link.
The [Tools](https://librecad.readthedocs.io/en/latest/ref/tools.html) section in the manual has different meaning.
The commands should be migrated from the wiki to the manual section [The Command Line](https://librecad.readthedocs.io/en/latest/guides/cmdline.html?highlight=commands#the-command-line) or possibly added as new reference in the manual appendix?

fa201, 2020-12-29:
From my point of view, having 2 places in the documentation listing commands (one in the reference chapters and one in command line / appendix) is not right for the same reason than one should not duplicate parts of code...

Before opening this issue I thought the same way but it is also very natural to have the commands just besides the explanation...

lordofbikes, 2020-12-29:
Of course, you are right @fa201, my fault, sorry for the confusion.
I missed the command column when I took a quick look at the tools table.
Maybe its time to get some glasses :rofl:

closed 2020-12-29 by lordofbikes

fa201, 2020-12-30:
@lordofbikes there is no offense :-)
Thanks for the PR.

commit aa5a427f (2020-12-29): changed Help->Online->Commands link, fixes #1215