LibreCAD/LibreCAD #1218: Can't export the lines color to svg

liufeijin, 2020-05-02:
when export to svg all line becomes black.

fa201, 2020-05-02:
Which version ?
Which system ?
How did you proceed ?

liufeijin, 2020-06-07:
it is ok

closed 2020-06-07 by liufeijin

aslze, 2020-06-08:
I just noticed this problem and then found this issue (closed??). This is the stable 2.1.3 (as installed by winget).

Exporting to **MakerCAM SVG** produces all black lines (`stroke="black"`). Like this:

```xml
<g lc:layername="xxx" lc:is_locked="false" lc:is_construction="false" fill="none" stroke="black" stroke-width="1">
```

The lines are originally several different colors.

Maybe this is a limitation of MakerCAM (which I know nothing about). But there is no other **SVG** export than that.

Is 2.2 RC stable enough to be used?

lordofbikes, 2020-06-08:
This fix is only in master branch, not in 2.3.1.
But the master is as stable as 2.3.1, so you can use nightly builds from our SourceForge downloads when you are on Windows or Mac.

liufeijin, 2020-06-09:
Yes when i use the newest version , i found it is ok. so i close it.

lordofbikes, 2020-06-09:
@liufeijin of course, my last post was to answer @aslze :grin: