LibreCAD/LibreCAD #1219: Appliocation is regularly crashing, seems to be random.

ekubovsky, 2020-05-03:
### Expected behavior
It is expected for application to not crash 

### Observed behavior
The application crashes randomly, no matter what action is performed. It crashes every 5 to 10 minutes, on different operations ranging from copying object to setting dimensions. No particular pattern has been observed.

### Steps to reproduce or sample file
Since it does it randomly, without a pattern - not sure how to reproduce

### Operating System and LibreCAD version info
**OS**:
* Win10 Pro, v. 10.0.18362 Build 18362

**LibreCAD**:
* Version: 2.2.0-rc1
* Compiler: GNU GCC 4.9.1
* Compiled on: Feb 16 2018
* Qt Version: 5.4.1
* Boost Version: 1.53.0

pyondaime, 2020-06-08:
Same issue happening here with both LibreCAD 2.1.3 and 2.2.0. It also happens with multiple files. I tried reinstalling the program multiple times, cleaning the registry, resetting the PC and still the issue persists.

**OS:**

> Windows 10 Home Version 1909 Build 18363.836

**LibreCAD:**

> Version: 2.2.0-rc1
> Compiler: GNU GCC 4.9.1
> Compiled on: Feb 16 2018
> Qt Version: 5.4.1
> Boost Version: 1.53.0
> System: Windows

sand1024, 2026-09-26:
outdated and reported for too old version

closed 2026-09-26 by sand1024