LibreCAD/LibreCAD #1225: complete the alias - instruction mapping

lordofbikes, 2020-05-22:
### Expected behavior
Each action which is available in menus or tool bars should be accessible from command line too
### Observed behavior
Not all actions have a command line instruction and therefor no aliases
 
### Steps to reproduce or sample file
`RS2::ActionDrawLineTangent1` is an example for an existing mapping
`RS2::ActionDrawLineTangent2` has a missing mapping
There are for sure more missing action/instruction mappings

The involved source files:
https://github.com/LibreCAD/LibreCAD/blob/574d9f0a73d4ed7f9612f222773de03b3c5ed5b9/librecad/src/lib/engine/rs.h#L162-L400
https://github.com/LibreCAD/LibreCAD/blob/574d9f0a73d4ed7f9612f222773de03b3c5ed5b9/librecad/src/cmd/rs_commands.cpp#L80-L563

### Operating System and LibreCAD version info
doesn't matter, latest master branch

From forum thread:
https://forum.librecad.org/Customize-shortcut-hotkeys-tp5718912.html

GSStnb, 2020-06-21:
I've gone through the file and added most of the missing commands.  As I did I noted that the existing commands don't have a common convention.  Is there any reason to establish one?  Or, not to?

The best reason to introduce a common convention is that the commands can be determined by the type of tool and the required parameters.  For example, all line tools commands will follow the convention starting with "line" and the "2p" - "line 2p" / "lin2p" / "l2p", the horizontal line tool - "linhor", "linver", etc.  

The convention I'm suggesting is as follows:
Category: (long - med - short form)
              line - lin - l
              circle - cir - c
              curve - arc - a
              ellipse - ell - e
              polyline - pl - p
              select - sel - s
              modify - mod - m
              (others as req'd)
  Parameters:
              point - p
              circle - c
              radius - r
              tangent - t
              angled - a
              vertical - v
              horizontal - h
              (others as req'd)

The biggest reason not to is that some current commands may change (which can be addressed in an alias file).

GSStnb, 2021-06-27:
Revisited with a more consistent implementation.  See PR #1331

jcbeppler, 2021-07-12:
@GSStnb May I suggest also to include the following?
- "m" to be also an alias to move as it is a command which is very common to use and the dynamic of clicking just "space bar"+"m"+"space bar" is quite handy;
- "cp" to be included for copy;

u2n, 2021-07-13:
^^^ 100% agree.

Pop-ups are indeed cumbersome; ok for learning but no good for regular use.

Perhaps a config setting could disable them, then the default action would be taken. Everything for speed.

GSStnb, 2021-07-15:
Moving an object doesn't require any command.  Selected object(s) can simply be dragged to a new position.  After making a selection, grab the selected object(s) close to the reference (anchor) point (it will snap to an endpoint, but careful NOT to grab an entity's handle) and drag it to a new position /target point (using snaps as needed/required).  Or use  ([Ctrl-[X] / [Ctrl]-[V]).  Either case, no pop-ups.

Ditto for copying / pasting ([Ctrl-[C] / [Ctrl]-[V]) - no pop-ups.

dxli, 2024-05-18:
Workaround exists

closed 2024-05-18 by dxli