LibreCAD/LibreCAD #1226: How does ttf2lff software convert ttf font to lff font?

lianzaozi, 2020-05-23:
How does ttf2lff software convert ttf font to lff font?
What is the specific operation of the software?

liufeijin, 2020-06-08:
ttf2lff   simsun.ttf  simsun.lff   in windows console (cmd)
But the ttf2lff.exe i too old maybe the lff font can't use, you need build the ttf2lff.exe first.

lordofbikes, 2021-04-04:
Most work is done by [FreeType](https://www.freetype.org/) library.
**ttf2lff** use FreeType to extract each font glyph from TTF file and convert the glyph into a LibreCAD friendly section in LFF file.
LibreCAD friendly means, that the glyphs in LFF contain only lines and can be rendered in LibreCAD as polyline.
See #1238 and https://github.com/LibreCAD/LibreCAD/issues/1252#issuecomment-650177740 for more information.

closed 2021-04-04 by lordofbikes