LibreCAD/LibreCAD #1228: Saving readonly files: no warning!

chbuka, 2020-05-27:
Trying to save to a file that is readonly

### Expected behavior
There should be a warning, that saving the file wasn't succesfull

### Observed behavior
No warning, no message. Librecad behaves like it has succesfull saved the file.

### Steps to reproduce or sample file
Open librecad, draw one thing, save it to a file.
Close librecad, set the file readonly (mode r--r--r--)
Open the file in librecad, draw another thing, click the save button, close the program
Open the file again: The second drawing was not saved!
### Operating System and LibreCAD version info
Debian 9.12 ( Paket: librecad; Version: 2.1.2-1+deb9u1)
LibreCAD Version: 2.1.2
Compiler: GNU GCC 6.3.0
Compiled on: May 19 2019
Qt Version: 5.7.1
Boost Version: 1.62.0
System: Debian GNU/Linux 9 (stretch)

fa201, 2020-05-29:
Thanks for reporting.

Bug confirmed on recent build:
Version: 2.2.0-rc1-201-g39300516
Compiler: GNU GCC 7.5.0
Compiled on: May 24 2020
Qt Version: 5.9.5
Boost Version: 1.65.1
System: Ubuntu 18.04.4 LTS

fa201, 2023-08-12:
Additional information:
1/ Before saving there is a warning message that the user should check the filename and permissions. This suggests that LibreCAD does not handle properly missing file or permission issue.
2/ Saving is shown as complete but actually is not.
[image]