LibreCAD/LibreCAD #1235: Reverse operation to explode?

Harvie, 2020-06-01:
Hello, i create rectangle, fillet the corners and i do offset. it is done correctly (all inside or outside the rectangle). But when i explode it, the offset is done for individual segments instead (some inside, some outside). Is there way to join exploded shape back to single object?

Before exploding:
[image]

After exploding:
[image]

Let's say i dont start with rectangle at all and i want to join individual segments to single loop and then do the correct offset on it.

fa201, 2020-06-01:
No the *join* feature is not implemented.

lordofbikes, 2020-06-01:
In LibreCAD _JOIN_ is called _Create Polyline from Existing Segments_.
Find it in the Tools -> Polyline menu or Polyline tool bar
[image]

This works only for lines and arcs.
An alternative is to create a block, therefore use the block list:
Select all entities and click _Create Block_

fa201, 2020-06-04:
Sorry I was unclear. You can create polyline as said by @lordofbikes but you cannot group entities (as in Office programs or other CAD software) whether they are not lines/arcs or whether they do not intersect.
So you cannot make sub-assemblies apart from blocks and polylines.

closed 2021-04-03 by DuncanLithgow

Harvie, 2021-04-04:
@DuncanLithgow Hello, why this was closed? Have you managed to solve the issue?

DuncanLithgow, 2021-04-04:
@fa201 and @lordofbikes might of course correct me, but as far as I can tell reading this thread the software is working as expected. You can create a block or a polyline. If you'd like to describe a feature request maybe this issue could be reopened as a feature request.

fa201, 2021-04-04:
If it is a feature request for group entities into one set which is not a block then it is already requested. [#1129 ](https://github.com/LibreCAD/LibreCAD/issues/1129)