LibreCAD/LibreCAD #1236: Draft mode warning

fa201, 2020-06-01:
Many new users have issues with the *Draft* mode which is not intuitive when one starts using LibreCAD.
I suggest that we implement a modal *Warning* window upon launching the command saying something like:
*Draft mode speeds up the view process but alter some view properties like line thickness, dimension values and hatches. If it is not you intent then you should cancel.*
We could also imagine adding a mention refering to User Manual.

dellus, 2020-06-01:
I have an even more radical proposal: why not drop Draft mode completely or at least hide it as a function to be activated explicitly, not in the the out of box appearance?
It seems to me to be an obsolete function. My experience is that with large drawings, which pose a challenge to LibreCAD 2 (LC 3 is aimed at tackling this), it doesn't help a lot.

fa201, 2020-06-03:
It also helps to switch to line width = 0. I think this is helpful. I am sure many people want this feature but I am also sure that many people use it without knowing it. So I suggest to warn everyone as a trade off.

qwasert-ben, 2025-01-06:
I know this is an old thread but I would like to weigh in.  I’m an x user of delta cad.  It only took me a first drawing and a few searches to figure out why my text and line thickness wasn’t showing up.  After checking that everything looked good in print preview I figured it was to help excelerate the program.  However, I do agree, that a pop up would be nice to toggle draft mode, espically for beginners.

sand1024, 2026-09-26:
@sand1024 change default value for draft mode to false