LibreCAD/LibreCAD #1238: Document introduction in lff font format

lianzaozi, 2020-06-02:
Are there any documents in lff format? I want to understand this format, but I can't find relevant information.

lordofbikes, 2021-04-04:
There is not much magic in LFF format.
Find a single line description in https://github.com/LibreCAD/LibreCAD/issues/1252#issuecomment-650177740, see first line in section **Interesting trick**. The trick also explains how to paste the letter coordinates to LibreCAD command line to visualize letters.

LFF is pure ASCII, inspecting any file and looking at the less than 400 code lines of [tools/ttf2lff/main.cpp](https://github.com/LibreCAD/LibreCAD/blob/master/tools/ttf2lff/main.cpp) should be a good base to understand the format.

closed 2021-04-04 by lordofbikes

app4soft, 2024-08-31:
Here is wiki with LFF font spec:

- https://github.com/LibreCAD/LibreCAD/wiki/LFF-Font-Format