LibreCAD/LibreCAD #1239: Support for DWG format

Helg1980, 2020-06-03:
Hi!
Please support the DWG format. There is a "open DWG" tab, but the function doesn't actually work.

liufeijin, 2020-06-08:
It will be crashed when i open DWG file.

XVilka, 2020-06-08:
See https://github.com/LibreCAD/LibreCAD/issues/1055

Helg1980, 2020-06-08:
... it is clear that nothing is clear...

bzuillsmith, 2020-08-15:
The program crashed for me on Windows 10 when attempting to open a DWG. No error message.

DuncanLithgow, 2020-12-28:
related to https://github.com/LibreCAD/LibreCAD_3/issues/207 on LibreCAD3

fr4ncks, 2020-12-28:
> related to LibreCAD/LibreCAD_3#207

Hi, I send this message to propose a solution, unfortunately not with
libreCAD, but to convert to DXF before opening, there is a soft (free for
non commercial use)from open design alliance that allows you to convert
files from DWG, it works perfect for me, even if it's not the right
solution (fun fact : I use it when AutoCAD refuse to register in DXF)

DuncanLithgow, 2020-12-28:
@fr4ncks this has been the approach of FreeCAD for a while. It works. But it cannot be bundled it must be downloaded manually. It is gratis, but only for as long as ODA wants it to be that way. As with Draftsight, Sketchup and now Fusion360 that can change at any time from gratis to purchase only.

lordofbikes, 2021-01-09:
No examples here, so no useful information for the issue.
The issue matches best with #988 and is tracked now there.

closed 2021-01-09 by lordofbikes