LibreCAD/LibreCAD #1243: I have converted 4 Chinese llf fonts for LibreCAD, but the size is bigger than 25m can't upload to github

liufeijin, 2020-06-08:
how can i upload the Chinese LLF fonts to LibreCAD ?
[image]
[image]

DuncanLithgow, 2021-04-03:
@lordofbikes can you take a look at this?

lordofbikes, 2021-04-04:
The LFF format is pure ASCII and the files should shrink significant in a ZIP archive.
You can post the ZIP here when the original licenses are valid for public sharing.
Best is when you also post original font file names with authors and license.
Because of the size I don't expect that we put them into the installer packages, but we can share them in our Resources repo and/or the wiki.

There may also be an issue with the filenames. I'm not sure what happens on different OS with UTF8 filenames.
The filename issue may be related with #1252 and #1319.

hsieb, 2023-03-31:
好兄弟能发我一份吗，中文字库
13207072760@163.com

cyannatoss, 2023-07-24:
好兄弟我也想要一份，之前找了半天只找到一個不能用的文泉驛字體
awsldnmd@gmail.com

dxli, 2023-07-24:
[WenQuanYi](http://wenq.org/) fonts are included in LibreCAD-2.1.3: [LibreCAD-2.1.3](https://github.com/LibreCAD/LibreCAD/archive/refs/tags/2.1.3.zip)

The WQY fonts are also avaialble at :+1: 
[WQY fonts](https://sourceforge.net/projects/librecad/files/Resources/fonts/)

DuncanLithgow, 2023-07-24:
That's fantastic!

Off topic: I don't know much about fonts, but I hope you will make them available through other well known channels/websites. Maybe https://fontlibrary.org/
\-------- Original Message --------

[WenQuanYi]: http://wenq.org/
[LibreCAD-2.1.3]: https://github.com/LibreCAD/LibreCAD/archive/refs/tags/2.1.3.zip
[WQY fonts]: https://sourceforge.net/projects/librecad/files/Resources/fonts/
[view it on GitHub]: https://github.com/LibreCAD/LibreCAD/issues/1243#issuecomment-1648126474
[unsubscribe]: https://github.com/
[AAYS35AHJINMYGP3HSINGKTXR2HOBA5CNFSM4NX2MTF2YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOMI6GUCQ.gif]: https://github.com/notifications/beacon/AAYS35AHJINMYGP3HSINGKTXR2HOBA5CNFSM4NX2MTF2YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOMI6GUCQ.gif

lordofbikes, 2023-08-08:
@DuncanLithgow because LibreCAD's LFF files are only usable with LibreCAD, it does not make much sense to provide them on any font platform.
Also LibreCAD fonts are usually converted from existing TTF fonts, which are already available on common platforms.

@dxli I've set up a wiki page to share not bundled fonts:
https://dokuwiki.librecad.org/doku.php/community:fonts
The WQY font can now be downloaded there.
It was removed basically because of it size. Because wqy-unicode.lff is over 41MB, when the remaining 46 fonts together are only about 26MB.

The plans are to extend the wiki page with more fonts and add a link inside the application to promote the fonts and to find them easier.
Probably we can move more fonts over there and remove them from deployment, to reduce our carbon footprint, every little helps.

Because there is no response from @liufeijin I close this now.

closed 2023-08-08 by lordofbikes