LibreCAD/LibreCAD #1244: Convert shx font

lianzaozi, 2020-06-08:
Is there any way or software that can convert shp/shx (shx files decompiled to get shx files) fonts to lff fonts?

liufeijin, 2020-06-08:
I think you need convert shx to tff then use ttf2lff convert ttf to lff

lianzaozi, 2020-06-08:
Do you know what conversion software can convert shx to ttf?

liufeijin, 2020-06-08:
Sorry, i don't find it still.

hsieb, 2023-04-12:
请问解决了吗，cad的单线矢量字体hztxt能在librecad上面显示吗

app4soft, 2024-08-18:
> Is there any way or software that can convert shp/shx (shx files decompiled to get shx files) fonts to lff fonts?

Yes. See:

- https://sourceforge.net/p/shx-font-converter/code/ci/master/tree/
    - https://github.com/Kharabadze/SHX_font_converter

@dxli @lordofbikes, Could it be merged to LibreCAD repo?

Also, on forum discussed some other way of SHX2LFF converting:

- https://forum.librecad.org/Another-Font-Conversion-Experiment-tp5723057.html
- https://forum.librecad.org/SHX-to-LFF-td5723053.html

lordofbikes, 2024-08-22:
> @dxli @lordofbikes, Could it be merged to LibreCAD repo?

I had a look at the repo and I don't want to merge this into LibreCAD's code base.
It looks like a quick'n'dirty hack, there are various issues which will surely raise CVE's in future.

This doesn't mean, that the converter does not work. It may work for what it is built and it may be fine as an external tool.
But to bundle it with LibreCAD it is too vulnerable, it might affect LibreCAD's trust and security rating.
The code would need a general refactoring or even a complete rewrite.

app4soft, 2024-08-22:
OK, there is SHX to Hershey Python lib: https://github.com/tatarize/shxparser

And source of Hershey to LFF `fontconverter` has been found (it might could be merged to `ttf2lff`, as derivated from it itself):

- https://github.com/LibreCAD/LibreCAD/issues/1245#issuecomment-2295078081

Gary-Wheeler, 2024-12-24:
Interesting point about keeping conversion programs separate from the main CAD program. 

CVE's being Common Vulnerabilities and Exposures.