LibreCAD/LibreCAD #1245: jhftolff conversion software

lianzaozi, 2020-06-08:
https://forum.librecad.org/LibreCAD-quot-lff-quot-fonts-continued-td4807481.html

I have written a converter from jhf (Hershey fonts) to lff
Is a qt4 console, you can get it:
http://sourceforge.net/projects/rallaz.u/files/fontconverter/fontconverter.zip/download

But the jhftolff conversion software mentioned in the link cannot be found.

lordofbikes, 2020-06-08:
This seems to be the work of an individual and was never part of LibreCAD.
The link from 2011 is broken, there is nothing we can do.

closed 2020-06-08 by lordofbikes

app4soft, 2024-08-18:
> > > http://sourceforge.net/projects/rallaz.u/files/fontconverter/fontconverter.zip/download
> >
> > But the jhftolff conversion software mentioned in the link cannot be found.
>
> This seems to be the work of an individual and was never part of LibreCAD.
> The link from 2011 is broken, there is nothing we can do.

[Guess](http://web.archive.org/web/20200912190715/https://github.com/whitequark/solvespace/issues/12#issuecomment-142200943), here is the source of `jhf2lff` (aka `fontconverter`):

-  https://github.com/Rallaz/LibreCAD/tree/master/tools/ttf2lff
    - https://github.com/Rallaz/LibreCAD/blob/master/tools/ttf2lff/lfftransform.cpp
        ```
        /****************************************************************************
        ** $Id:  main.cpp
        ** fontconverter a qt console program to convert .jhf to .lff fonts
        **
        ** Copyright (C) 2011 -Rallaz. <***@***>
        **
        ** This program is free software; you can redistribute it and/or
        ** modify it under the terms of the GNU General Public Licence
        ** as published by the Free Software Foundation; either version 3
        ** of the License, or (at your option) any later option.
        ```

@dxli @lordofbikes, could it be merged back to LibreCAD repo? (Rallaz might be mistakenly mentioned GPLv3, insted of GPLv2)

Also, there is a Python implementation of some basic JHF to LFF:

- https://www.github.com/solvespace/solvespace/issues/1042#issuecomment-1842222488
    > I wrote [jhftool.py](https://github.com/solvespace/solvespace/files/13577502/jhftool.txt). License: public domain.

It is designed especially for converting [Mingti-basic.hf.txt](https://github.com/LingDong-/chinese-hershey-font/blob/master/dist/hershey/Mingti-basic.hf.txt) (CJK URO and Ext-A) to LFF.

(backup copy of `jhftool.py`: [jhftool.txt](https://github.com/user-attachments/files/16647025/jhftool.txt))