LibreCAD/LibreCAD #1246: LibreCAD can't open DWG file(from autocad), Autocad can't read dwg made by LibreCAD

liufeijin, 2020-06-08:
So, if LibreCAD2 doesn't change to LibreDWG, kindly please  stop LibreCAD2 and develop LibreCAD3 only.
Hope LibreCAD3 can be released the first revision ASAP.

zhengf312, 2020-06-20:
我也是一名LibreCAD的爱好者，最近在做一个学校的项目，能留个联系方式，请教你一些问题吗

zhengf312, 2020-06-20:
812638322@qq.com  这是我的邮箱，可以将联系方式发到我的邮箱，我做LibreCAD已经很久了，最近也在考虑将libredxf换成LibreDWG，联系RvT，也没回复我，计划自己写了

lordofbikes, 2020-06-20:
Everything about license issues, which prohibit the use of LibreDWG in LibreCAD, is already said in #1055 

@zhengf312 you are welcome to join our discussions at https://librecad.zulipchat.com/.

closed 2020-06-20 by lordofbikes

liufeijin, 2020-06-20:
As an end user, in fact,i am not care on which lib is better. I just want LibreCAD2 can read and write DWG file. If someone can fix the problem on LibreCAD2 , it will be better.